jopohl/urh #1051: Freezes & Crashes

TheBarret, 2023-05-05:
This application is not stable, freezes or crashes a lot while using it in any capacity, doesn't seem your using multithreading.

other then that, great idea, would love to see it becoming better!

andynoack, 2023-05-06:
Dude, you are guessing that the application doesn't use multithreading while posting an error log that excepts in mutliprocessing...
Anyway, I'll try to help and created a branch [fix_rtlsdrtcp](https://github.com/jopohl/urh/tree/fix_rtlsdrtcp) for you. Please check it out!

TheBarret, 2023-05-07:
> Dude, you are guessing that the application doesn't use multithreading while posting an error log that excepts in mutliprocessing... Anyway, I'll try to help and created a branch [fix_rtlsdrtcp](https://github.com/jopohl/urh/tree/fix_rtlsdrtcp) for you. Please check it out!

i did say 'seems' that assumption was made because the UI thread is visibly dropping frames, if im wrong then im wrong ofc.

andynoack, 2023-05-09:
I would like to close this issue. Can you confirm that the fix works for you?

TheBarret, 2023-05-09:
> I would like to close this issue. Can you confirm that the fix works for you?

Thats all up to you Mr. Andy, I'm only reporting it and that's where my part stops.

andynoack, 2023-05-09:
That is not how it works! If you don't want to test the bugfix that I have written for you personally, it is up to you (but rude of course). 
I don't have your hardware to test it, so I'll delete the branch soon (within one week). If you want to give constructive feedback or if you are interested in solving your problem, reach out to me.

closed 2023-05-09 by andynoack

TheBarret, 2026-03-29:
> wait how did you implement the fix?

I wish, but the maintainer behaves like a dick.