jopohl/urh #1053: URH 2.9.4 and Airspy Mini - need more gain. Would appreciate more bandwidth variety.

Bigblock, 2023-05-24:
I'm using URH 2.9.4 on a Macbook Pro 2019, 64 GB RAM, AMD Radeon 5500M.

Used a Nooelec NESDR Smart before and bought an Airspy Mini now, because i needed more bandwidth.

With the NESDR Smart i had no gain problems at all and could vary the gain to my needs.

With the Airspy Mini gain is much to low, even if i turn the gain up to 15. So the signals i record are not usable.

I don't know if i'm doing something wrong here, but i would expect that the gain range should be the same.

Also i would like to vary the bandwidth, which seems to be fixed to 6 MHz. So if i could to let's say 2 and 4 MHz that would speed up my workflow extremely.

Thanks in advance.

andynoack, 2023-05-24:
Hey, as noted in urh/dev/config.py, the AirSpy Mini devices use pretty fixed values for sample rate, bandwidth and also gain. If you like, you can play around with these constants but be aware that they should already match with your hardware. We are only able to provide settings that the hardware respectively the driver supports.

Bigblock, 2023-05-25:
Hi andynoack,

thanks for your fast response. And thanks to the urh/dev/config.py link. Unfortunately, i haven't read that before.

So what is a hands-on solution here, if the driver is already "maxed out"?

An additional hf preamp to power up the input-signal?

Or just kick the Airspy Mini and get something else like a SDRPlay or LimeSDR?

At the end of the day, i need a better sample rate and wider bandwidth. The RTL-SDR solution i had before is too limited for my needs.

What would you recommend? BTW a BladeRF is out of reach. ;)

andynoack, 2023-05-25:
Hi @Bigblock 
well, in our experience input gain isn't the problem in most cases. On the right side of your recorded signals in interpretation there is a y-zoom slider that makes working with low gain signals easier. Often the best solution is to move the SDR-Antenna closer to the sender.
Note that with higher gain comes higher noise, so the better SNR you are searching for could not be reached.

The best TX capable SDR is in our experience the HackRF. With much (!) more money, I would go for an USRP N-series, whereby these devices are actively cooled (i.e. loud) and not portable. In many of our test cases the HackRFs precision was good enough, so this is our first grab.

Bigblock, 2023-05-25:
Hi @andynoack 

thanks for your advice. Luckily, i ordered the Airspy Mini from the official german distributor WIMO. A quick phone call and they take the Airspy Mini back and sent me an original Great Scott Hack RF One – well, for a little bit more money. ;)

The USRP N-series is unfortunately not in my budget range.

Thanks again and if you don't mind i'll report my experiences with the Hack RF One.