jopohl/urh #1054: URH crashes on RX with HackRF on Debian

pielgrzym, 2023-05-27:
##### Expected Behavior

Run any RX related action - like spectrum analyzer or record signal.

##### Actual Behavior

No RX is started, following error is in log:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.11/multiprocessing/spawn.py", line 120, in spawn_main
    exitcode = _main(fd, parent_sentinel)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/multiprocessing/spawn.py", line 130, in _main
    self = reduction.pickle.load(from_parent)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pielgrzym/tmp/urh/src/urh/dev/native/HackRF.py", line 7, in <module>
    from urh.dev.native.Device import Device
  File "/home/pielgrzym/tmp/urh/src/urh/dev/native/Device.py", line 13, in <module>
    from urh.dev.native.SendConfig import SendConfig
  File "/home/pielgrzym/tmp/urh/src/urh/dev/native/SendConfig.py", line 5, in <module>
    from urh.util.RingBuffer import RingBuffer
  File "/home/pielgrzym/tmp/urh/src/urh/util/RingBuffer.py", line 4, in <module>
    from urh.signalprocessing.IQArray import IQArray
  File "/home/pielgrzym/tmp/urh/src/urh/signalprocessing/IQArray.py", line 8, in <module>
    from urh.cythonext.util import get_magnitudes
ModuleNotFoundError: No module named 'urh.cythonext.util'
```

##### Platform Specifications

 - OS: Debian Testing
 - URH version: 2.9.4 and git HEAD
 - Python version: 3.11.2
 - Installed via pip + virtualenv

andynoack, 2023-05-29:
Try rebuilding the cython extensions with python build.py in src/urh/cythonext. It seems to me that URH cannot find the extenstions (ModuleNotFoundError: No module named 'urh.cythonext.util').

andynoack, 2023-07-12:
Closed due to inactivity.

closed 2023-07-12 by andynoack