jopohl/urh #1061: Error installing on Raspberry pi 4 with rtl-sdr

SimoSera, 2023-07-26:
When I try installing URH with "sudo python3 -m pip install urh" i get an error. 
The first time I tried installing I was missing Cython, after installing it I tried again and now I get another error that I don't understand.

This is the output i get when i run the command to install URH:

```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting urh
  Using cached urh-2.9.4.tar.gz (3.0 MB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [49 lines of output]
      warning: src/urh/dev/native/lib/rtlsdr.pyx:17:0: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
      warning: src/urh/dev/native/lib/crtlsdr.pxd:33:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
      
      Error compiling Cython file:
      ------------------------------------------------------------
      ...
          read_bytes_async
          :return: 0 on success
          """
          global f
          f = callback
          return crtlsdr.rtlsdr_read_async(_c_device, _c_callback_recv, <void *>connection, 0, 0)
                                                      ^
      ------------------------------------------------------------
      
      src/urh/dev/native/lib/rtlsdr.pyx:326:48: Cannot assign type 'void (unsigned char *, uint32_t, void *) except *' to 'rtlsdr_read_async_cb_t'
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-16wia5cu/urh_72a96fe164e04b7e809732d360d7e1ec/setup.py", line 138, in <module>
          ext_modules=get_extensions(),
        File "/tmp/pip-install-16wia5cu/urh_72a96fe164e04b7e809732d360d7e1ec/setup.py", line 97, in get_extensions
          extensions = cythonize(extensions, compiler_directives=COMPILER_DIRECTIVES, compile_time_env=device_extras)
        File "/usr/local/lib/python3.9/dist-packages/Cython/Build/Dependencies.py", line 1134, in cythonize
          cythonize_one(*args)
        File "/usr/local/lib/python3.9/dist-packages/Cython/Build/Dependencies.py", line 1301, in cythonize_one
          raise CompileError(None, pyx_file)
      Cython.Compiler.Errors.CompileError: src/urh/dev/native/lib/rtlsdr.pyx
      Skipping native support for airspy
      Skipping native support for bladerf
      Skipping native support for hackrf
      Skipping native support for limesdr
      Skipping native support for plutosdr
      Found rtlsdr lib. Will compile with native rtlsdr support
      Skipping RTLSDR_BANDWIDTH_SUPPORT as installed driver does not support it
      Skipping native support for usrp
      Skipping native support for sdrplay
      Compiling src/urh/cythonext/auto_interpretation.pyx because it changed.
      Compiling src/urh/cythonext/util.pyx because it changed.
      Compiling src/urh/cythonext/path_creator.pyx because it changed.
      Compiling src/urh/cythonext/awre_util.pyx because it changed.
      Compiling src/urh/cythonext/signal_functions.pyx because it changed.
      Compiling src/urh/dev/native/lib/rtlsdr.pyx because it changed.
      [1/6] Cythonizing src/urh/cythonext/auto_interpretation.pyx
      [2/6] Cythonizing src/urh/cythonext/awre_util.pyx
      [3/6] Cythonizing src/urh/cythonext/path_creator.pyx
      [4/6] Cythonizing src/urh/cythonext/signal_functions.pyx
      [5/6] Cythonizing src/urh/cythonext/util.pyx
      [6/6] Cythonizing src/urh/dev/native/lib/rtlsdr.pyx
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

```

I am running a Raspberry pi 4 model B with 4GB of ram with Raspberry Pi OS.
I am also using a RTL-SDR and i already have the drivers installed, I have installed gqrxsdr and rtl-433 and they run fine.
pip --version : pip 23.2.1 from /usr/local/lib/python3.9/dist-packages/pip (python 3.9)
python --version : Python 3.9.2

andynoack, 2023-07-27:
Well, officially there is no Raspberry Pi support but it seems, that most of the critical stuff seems to compile, asides from rtlsdr.pyx. The problem probably is a type problem in line 326, maybe someone knows how to make this ARM v8 compatible? I don't have a RPi4 nor a RTLSDR at my workplace, so I call for PRs.

SimoSera, 2023-07-27:
I resolved by commenting the part the gave the error. 
In the file rtlsdr.pyx  in urh/src/urh/dev/native/lib I edited this part:

```
cpdef int read_async(callback, connection):
    """
    Read samples from the device asynchronously. This function will block until
    it is being canceled using rtlsdr_cancel_async()
    read_bytes_async
    :return: 0 on success
    """
    global f
    f = callback
    return crtlsdr.rtlsdr_read_async(_c_device, _c_callback_recv, <void *>connection, 0, 0)

```

to :

```
"""
cpdef int read_async(callback, connection):

    Read samples from the device asynchronously. This function will block until
    it is being canceled using rtlsdr_cancel_async()
    read_bytes_async
    :return: 0 on success

    global f
    f = callback
    return crtlsdr.rtlsdr_read_async(_c_device, _c_callback_recv, <void *>connection, 0, 0)
"""
```

This way I commented the code and when installing the error disappeared.
For now urh seems to work fine, I only tried the spectrum analyzer and creating a project.

closed 2023-07-27 by SimoSera