jopohl/urh #1064: You need Cython to build URH's extensions!

tomcass240, 2023-09-09:
Describe the bug
Trying to install urh using pipx install urh. I get the message

Fatal error from pip prevented installation. Full pip output in file:
/home/z/.local/pipx/logs/cmd_2023-09-06_22.54.24_pip_errors.log

pip seemed to fail to build package:
urh

Some possibly relevant errors from pip install:
error: subprocess-exited-with-error
error: metadata-generation-failed

cat /root/.local/pipx/logs/cmd_2023-09-06_22.51.35_pip_errors.log
PIP STDOUT
----------
Collecting urh
  Using cached urh-2.9.4.tar.gz (3.0 MB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'

PIP STDERR
----------
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [2 lines of output]
      You need Cython to build URH's extensions!
      You can get it e.g. with python3 -m pip install cython.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

How to reproduce
Try to install urh with pipx on Debian. Python3 and Cython3 both installed using package manager.

Expected behavior
Installed correctly as Cython is installed.

If I try typing sudo pip3 install urh

I get this error message:

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

I found advice on here suggesting using venv and installing it to a python virtual environment. This worked but it didn't have access to any of my existing libraries so there was no device support with it, making it useless.

I've just uninstalled it for now. Are there any better programs out there than URH?

jopohl, 2023-09-09:
You run Python 3.11 for which there are no prebuilt wheels on PyPi.

You have couple of options:

1. Use snap
2. Use the URH docker image
3. install the SDR dev packages you want to use e.g. `apt install libhackrf-dev` and, afterwards, install URH again

jopohl, 2023-10-28:
Just released v2.9.5 which includes Python wheels for 3.10, 3.11 and 3.12. So installation with pip / pipx should be smooth now for recent Python versions.

closed 2023-10-28 by jopohl

Crsarmv7l, 2024-10-29:
I am having this exact issue with python 3.12, Cython was installed via pip and installed failed citing no cython. I installed cython via apt, same result

arrase, 2025-01-02:
+1 Same here.

arrase, 2025-01-02:
For Raspberry pi OS:

> 1. Use snap

error: snap "urh" is not available on stable for this architecture (arm64) but exists on other architectures (amd64)

arrase, 2025-01-02:
@Crsarmv7l 

> I am having this exact issue with python 3.12, Cython was installed via pip and installed failed citing no cython. I installed cython via apt, same result

After install cython3 with apt run:

```
pipx install urh --system-site-packages
```