jopohl/urh #1067: RTL-TCP is setting bias tee to the same value as sample rate

oddstr13, 2023-10-04:
##### Expected Behavior
I would expect bias tee to be a separate option, and not tied to the sample rate setting.

##### Actual Behavior
RTL-TCP is setting bias tee to the same value as sample rate.

I am not sure of what the actual consequence of this is.

##### Steps To Reproduce

1. Connect to RTL-TCP
2. Observe in RTL-TCPs logs that bias tee is set to the same value as sample rate

##### Screenshots

[image]

```
client accepted! workstation.local 43600
Allocating 15 zero-copy buffers
set freq 442169575
set sample rate 240000
set bias tee 240000
set freq correction 0
set direct sampling 0
Disabled direct sampling mode
set gain 28
worker socket bye
Signal caught, exiting!
comm recv bye
Signal caught, exiting!
all threads dead..
listening...
Use the device argument 'rtl_tcp=0.0.0.0:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
```

##### Platform Specifications

 - OS: Linux Mint
 - URH version: 2.9.4
 - Python version: 3.10
 - Installed via pipx

andynoack, 2023-10-06:
BiasTee is a binary value (Enabled/Disabled), so setting it to a non-zero value will enable it.

oddstr13, 2023-10-06:
As far as I can tell, RTL-TCP doesn't have support for bias tee in the UI, so I wouldn't expect the rtl-tcp server to receive commands to set it in the first place.

I humbly request that this issue be re-opened, as it does seem like something is not working as intended here.

closed 2025-06-03 by andynoack