jopohl/urh #1069: ADALM-PLUTO SDR is not recognized

ahuri, 2023-10-17:
Hi.

I tried to capture a signal using Pluto-SDR but it's not recognized by URH.
The driver is up-to-date.

##### Steps To Reproduce
1. Go to 'File'
2. Click on 'Record signal...'
3. Choose 'PlutoSDR' under "Device".
4. Click 'Start' to start capturing.

##### Screenshots
[image]

##### Platform Specifications

 - OS: Windows 10
 - URH version: 2.9.4
 - Python version: [e.g. 3.7]
 - Installed via [e.g. pip]

andynoack, 2023-10-17:
Could you test whether your device is recognized by URH under Linux?

ahuri, 2023-10-18:
I'm on it.
Will update soon

ahuri, 2023-10-18:
OK.
Finally installed Ubuntu 22.04.3
1. I tried to install URH using snapd - it failed to launch it (because of a Qt issue I could not figure out how to fix).
2. I tried to install it using pip3 - but did not understand how to launch it.
3. I cloned the GitHub repository and executed main.py - but no device is listed but "RTL-TCP" in the recording signal window.

For now, I'm still struggling to make it work.
If you have any solution - please let me know.

ahuri, 2023-10-18:
OK. Seams like the snapd installation had permissions issues, "sudo urh" worked and the PlutoSDR is recognized!
Have not idea why in Windows it does not work.

andynoack, 2023-10-18:
Hey, now we are certain that your device is working, that is a plus. Probably it is a driver issue under Windows then. Have you tried to completely remove the Pluto (and in a second step also the related) drivers and install them News?

ahuri, 2023-10-18:
Yes.
Just for clarification, the Pluto seems to work in SDR# for instance (in Windows).

andynoack, 2023-10-18:
That is good! I had a look in our Implementation, maybe our backend is not compatible with your driver. Maybe an older or never (guess not) works. If yes, please let us know which driver works and which does not.

ahuri, 2023-10-19:
I don't think I've ever installed a driver for that SDR...
Just plug and play.

Oleg-cbrRR, 2023-11-01:
under windows you just have to specify DEVICE IDENFIFIER to "ip:192.168.2.1"

luzpaz, 2024-02-02:
Any progress here ?

andynoack, 2024-02-02:
The issue seems to be resolved since November 1st. Thank you, Oleg!

closed 2024-02-02 by andynoack