jopohl/urh #1070: Saving as Flipper Zero format (.sub) seems to be wrong

ahuri, 2023-10-19:
##### Expected Behavior
I recorded a signal that took less than a second send and saved it as Flipper Zero format (.sub).
Expected to use the Flipper device for sending this signal.

##### Actual Behavior
A very long file was created, hundreds of lines.
It took more than a minute to Flipper to send its content, until I stopped it before ending.

##### Steps To Reproduce
1. Go to "Record Signal" window.
2. Record a signal.
3. Save it with ".sub" extension.
4. Copy the file to Flipper Zero device and try to broadcast it.

##### Screenshots
The original signal:
[image]

The signal Flipper sends after saving the original signal as ".sub" format (stopped before ending):
[image]

##### Platform Specifications
 - OS: Ubuntu 22.04.3
 - URH version: 2.9.4
 - Python version: 3.10.12
 - Installed via pip3

andynoack, 2023-10-19:
Hey @ahuri,
saving as .sub does only work under certain conditions. Please use the new Export button in the generator, this should produce better results.
However, note that using the .sub format or the Flipper in general means a dramatic signal quality loss (precision is only 1 bit!) because the Flipper is NOT exactly a SDR.
Recently I also discovered some problems with a Flipper replay (maybe my setup or use case...), so please keep me updated.

ahuri, 2023-10-20:
Until now I have not managed to emulate a replay attack successfully using the Flipper.
I used an SDR to track the signal emulation of the Flipper in comparison to the original signal; they look very different.
The idea was to capture a signal using URH, save it in the Flipper, and try to send it.
But as I mentioned, it does not work.

godmar, 2024-07-26:
I recorded a signal at 906.40MHz, but the exported .sub file contained these lines:
```
Filetype: Flipper SubGhz RAW File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: RAW
```
which has the wrong frequency.
Also, the original recording was 5.5MB big, the .sub file only 4.7KB.

> Please use the new Export button in the generator, this should produce better results.

I installed urh via pip and don't see an export button there.

godmar, 2024-07-26:
I think I know the answer to my question: https://github.com/jopohl/urh/issues/989#issuecomment-1218792851

smolbun, 2025-10-19:
Tried capturing some OOK signals from my remote, when imported back to the pulse plotter there were no pauses between each signals that were sent. 

For my case each signal should have a 10ms pause between them but they were all concatenated with one another, it does not respect the pause value listed beside the protocols tab.