jopohl/urh #1075: Trojan:Win32/Sabsik.FL.A!ml Detected in x64

yasbean, 2023-10-28:
##### Expected Behavior
Install and enjoy!

##### Actual Behavior
I installed the new v2.9.5 and when I opened URH, Windows 11 warned that urh.exe is infected with Trojan:Win32/Sabsik.FL.A!ml

##### Steps To Reproduce

1. Download Universal.Radio.Hacker-2.9.5-x64.exe to Windows 11
2. Open URH
3. Windows security warned "Detected: Trojan:Win32/Sabsik.FL.A!ml"
##### Screenshots

##### Platform Specifications

 - OS: Windows 11
 - URH version: 2.9.5
 - Python version: ?
 - Installed via GitHub installation package

jopohl, 2023-10-28:
Most likely a false positive. I just uploaded the installer to Virustotal and it reported no hits:
[image]

See also my [comment](https://github.com/jopohl/urh/issues/921#issuecomment-992187425) here on that matter - please report this to your antivirus vendor.

yasbean, 2023-10-28:
Thank you for the link to the other report.  I am not sure how to report this to the huge dark hole that is Microsoft so I will probably just go over to Linux where I should have been doing this.  BTW, here is the result after opening URH:

[image]

jopohl, 2023-10-28:
I just reported the installer file [here](https://www.microsoft.com/en-us/wdsi/filesubmission) as misclassified malware and linked this issue for reference. Let's hope Microsoft will review this soon. 

I will close this issue as there is nothing more we can do from our side.

closed 2023-10-28 by jopohl

jopohl, 2023-10-29:
Here is the response from Microsoft:

> We cannot reproduce any detection on the file. If the detection is still observed, follow the steps below to capture support log files from the system reporting detection.
>
> From an elevated command prompt, change to directory "%programfiles%\windows defender" and execute mpcmdrun.exe with option GetFiles:
cd "%programfiles%\windows defender"
mpcmdrun.exe -GetFiles
>
> All created log files will be compressed into MPSupportFiles.cab. Please send us the detected file and MPSupportFiles.cab using https://aka.ms/wdsi. We will continue the investigation once we receive the support log files.
>
> Thank you for contacting Microsoft.

Could you please run Windows update and check whether the installer is still marked as malicious afterwards? If so, I appreciate if you could send them the support log files as outlined in their answer.

yasbean, 2023-10-29:
After a full system update, I have scanned the same installation file this morning and no malware was detected.  It appears that your report helped them correct the problem.  Thank you for the wonderful software!

yasbean, 2023-11-01:
Windows Defender again flagged URH as Trojan Malware.  I have submitted the log files to Microsoft.

yasbean, 2023-11-01:
Windows Defender again flagged URH as Trojan Malware.  I have submitted the log files to Microsoft but  and will wait for a reply.  Now, Windows keeps quarantining the application.

yasbean, 2023-11-15:
After reporting to Microsoft, I did not receive a response, but after re-installing and scanning several times, it has not been flagged, so I guess they have fixed the problem.

jopohl, 2023-11-16:
Awesome, let's hope they fixed it for good this time. Thanks for getting in touch with Microsoft and providing them with the logs.