jopohl/urh #1078: adjustable noise center

Mofef, 2023-10-30:
##### Is your feature request related to a problem?
I was wondering why the noise amplitude can be adjusted but not the position of the center of the noise. otherwise for bad SNR one half of the signal might be detected as noise. 
Am I missing something? Or maybe what i see (see screenshots) should never happen, if the tool is used correctly?

##### Describe the solution you'd like
between noise and center there should be a noise center adjustment

##### Describe alternatives you've considered
i'm missing the experience to come up with alternatives. sorry :/

##### Additional context
Sady i can't find the specific signal where i had this issue. but basically I see that the center of the noise range is not where the 0 level of the analog view is. this is what i would liek to adjust:
[image]

andynoack, 2023-10-31:
Hey Moritz,

the important part of a signal is of course the data. You need a solid SNR to differentiate the data from noise and that is where URHs noise filter comes into play. Usually data and noise have the same center unless you do not have more than one channel, so there is no need to set a second center for the noise. In case you have more than one channel, you can extract single channels using the Spectogram view.
If you like to discuss such things with us, please join our slack channel. Sometimes a dialog helps!
Please close the issue if your question is solved.

Best regards
Andreas

Mofef, 2023-10-31:
Sounds good, i will say hi in the slack after work!

Though I think there is a small misunderstanding.: 
>  Usually data and noise have the same center 

agreed!

>  so there is no need to set a second center for the noise

maybe better to say: a center for the analog signal (amplitude over time) as opposed to the center of the demodulated signal (frequency over time) does that make sense?

jopohl, 2023-11-01:
For the noise it does not really make sense to have a configurable center although I get where you are coming from. Let me give you a short break-down of how URH handles noise behind the scenes to make this more clear.

1. URH calculates the __magnitude__ of the signal i.e. we end up with values >= 0 over time
2. Every sample with a magnitude lower than the configured `noise_threshold` is considered noise and it's magnitude set to 0. You can visualize the noise threshold as a straight line that goes from left to right through the magnitudes.
3. Draw the original signal after cutting off noise

Note, the `noise_threshold` is not used at all in the analog view where it is drawn that's probably the source of confusion: The red noise area we draw in URH is just an approximate visualization.
The correct visualization would be to draw the magnitudes over time as outlined in point 2 and let the noise threshold go from left to right.

Mofef, 2023-11-09:
Hi, 

Thanks for the very helpful post! 
You guessed exactly right that the visualization was throwing me off.

Also thanks for the advice.
I didn't know there is way to apply a bandpass filter! I still coudn't find it untill I googled and found it mentioned in the URH paper... I have to admit i didn't read it 😔
Regarding the straight line: I panned to an are where there is no signal to visualize how the 0-line is claerly not in the center of the noise. I understand now that that doesn't mean anything

closed 2023-11-09 by Mofef