jopohl/urh #1085: 2 MHz samplerate signal not recognized

Botox200, 2023-11-10:
##### Expected Behavior
a 2 MHz samplerate recorded signal should be reopened and treated like a 2 MHz signal. By default the new-opened 2 MHz samplerate signal is always interpreted as a 1 MHz samplerate signal. 

##### Actual Behavior
recording a 2 MHz samplerate signal with a rtl-sdr and reopen it leads to a 1 MHz interpreted signal. 
recording a 1 MHz samplerate signal with a rtl-sdr and reopen it leads to a 1 MHz interpreted signal.

Changing the settings in the options does not resist. Changing multiple devices in the menu like HackRF and SdrPlay confuses the record samplerate settings.

##### Steps To Reproduce
1. Record a rtl-sdr signal with a 2 Mhz samplerate
2. Save the signal
3. Load the signal
4. Decode it
5. Send it with RfCat or FlipperZero
6. Incorrect signal

Change the signal in 1. to a 1 MHz samplerate will decode a successfull signal.

##### Platform Specifications
Kubuntu 22.04

btw. excellent program!!!

andynoack, 2023-11-10:
Unfortunately there is no place in .complex files (de facto standard) to save the frequency, so we cannot do much about it...sorry!

closed 2023-11-10 by andynoack

Botox200, 2023-11-10:
Ok. Thank you.