jopohl/urh #1086: issue with fsk demodulation

heneault, 2023-11-10:
##### Expected Behavior
demodulation should detect a series of 01010101 as highlighted in the image

##### Actual Behavior
demodulation only detect 00...000 in FSK

Samples/symbol seems to be ok. Filtering was applied. 

##### Screenshots
[image]

jopohl, 2023-11-12:
Did you try increasing the error tolerance?

andynoack, 2023-11-12:
Actually your signal does not look clean at all. I assume that you see many fallbacks to the center if you zoom in far enough. As @jopohl said, you either have to increase the error tolerance or apply a better filter or record the signal in a more clean way.

heneault, 2023-11-15:
got it I understand what you meant. I used the band pass filter and it fixed it. Thank you for your help.

closed 2023-11-15 by heneault