jopohl/urh #1087: Support Cython 3.0+

befeleme, 2023-11-13:
In Fedora Linux, we've successfully built `urh` with Cython 3.0.5 (see [build log](https://download.copr.fedorainfracloud.org/results/churchyard/cython-3-forced/fedora-rawhide-x86_64/06624196-urh/builder-live.log.gz) for details).
Would you consider unpinning the requirement on `Cython < 3` in the project? I can send the PR.

jopohl, 2023-11-13:
Yes, please send the PR.

closed 2024-01-21 by jopohl

parkerlreed, 2025-01-10:
Is this not working?

```
[parker@parker-framework build]$ python -m venv urh
[parker@parker-framework build]$ source urh/bin/activate
(urh) [parker@parker-framework build]$ pip install cython
Looking in indexes: https://pypi.org/simple, https://TybWQGxonh4y2gzDSqNJ:****@pypi.fury.io/simberobotics/
Collecting cython
  Downloading Cython-3.0.11-cp313-cp313-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (3.2 kB)
Downloading Cython-3.0.11-cp313-cp313-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (3.5 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.5/3.5 MB 18.3 MB/s eta 0:00:00
Installing collected packages: cython
Successfully installed cython-3.0.11
(urh) [parker@parker-framework build]$ pip install urh
Looking in indexes: https://pypi.org/simple, https://TybWQGxonh4y2gzDSqNJ:****@pypi.fury.io/simberobotics/
Collecting urh
  Using cached urh-2.9.8.tar.gz (3.3 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [2 lines of output]
      You need Cython to build URH's extensions!
      You can get it e.g. with python3 -m pip install cython.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

parkerlreed, 2025-01-10:
You need setuptools despite it not being a default anymore. Would be nice to have *that* as he error message.

PR #1099 (2024-01-21): Add support for Cython3
fix #1087