jopohl/urh #1092: Bits are not getting properly decoded in URH

JasleenKaur2308, 2023-12-06:
I am trying to decode the bit for 4 fsk for 4 different deviations , for positive deviation ( +648 Hz, +1994 Hz ) my bits are correctly decoding with respective symbols (+648 Hz = 00 and +1994 Hz = 01) but for negative deviation (- 648 Hz , - 1994 Hz) my bits are not correctly decoding . what settings i need to change ?

andynoack, 2023-12-06:
Hey, you could apply a custom decoding using the substitution block. There you are able to replace the decoded bits from URH with a decoding you prefer.

JasleenKaur2308, 2023-12-06:
I have attached some screenshots also , for positive values its coming correct but its wrong for negative value . According to symbol mapping it should be (+648Hz =00 , +1944 Hz =01 , -1944 Hz =11, -648 Hz = -648 Hz ). Only positive values are decoding properly . 
[image]
[image]

andynoack, 2023-12-06:
In demodulated view you are able to move the three centers for 2bit/symbol signals. Please check that they are set properly.

JasleenKaur2308, 2023-12-07:
Thank you so much @andynoack , now im able to decode the bits.

andynoack, 2023-12-11:
Great to hear that!

closed 2023-12-11 by andynoack