jopohl/urh #1094: Zynonym trojan/virus in Version 2.9.5?

WB0OCV, 2023-12-27:
##### Expected Behavior

##### Actual Behavior

##### Steps To Reproduce

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

##### Screenshots

##### Platform Specifications

 - OS: [e.g. Arch Linux]
 - URH version: [e.g. 2.4.2]
 - Python version: [e.g. 3.7]
 - Installed via [e.g. pip]

WB0OCV, 2023-12-27:
I just downloaded the 64-bit version 2.9.5 (28-Oct-2023), and upon installation, Windows Defender quarantines the urh_debug.exe executable as containing the Zynonym trojan/virus. I tried downloading the 32-bit version of v2.9.5, and Google blocks the download entirely as being a dangerous file. I downloaded and installed the 64-bit version of v2.9.4 (28-Jan-2023), and it installs without any problem or identification of a virus. Can the author of URH v2.9.5 please check the program for potential incorporation of a Trojan/virus and recomile if necessary? Thank you.

jopohl, 2023-12-27:
Looks like a duplicate of https://github.com/jopohl/urh/issues/1075

jopohl, 2024-01-14:
The false-positives are caused by PyInstaller - see [here](https://github.com/pyinstaller/pyinstaller/issues?q=label%3Aantivirus-false-positives+) for a list of issues.

Just released URH v2.9.6 with a more recent Python version (3.11) and built with PyInstaller 6.3.0.

Please contact your anti-virus vendor if the new version gets still quarantined. There is not much URH can do about such false-positives.

closed 2024-01-14 by jopohl