jopohl/urh #1095: Play .wav created from AOR receiver.

Yorch23, 2024-01-04:
It would be possible to add playback in HackRF from URH of .wav files created with AOR receiver with the following characteristics (Not editable in aor):
-30kHz Bandwidth (+/-15kHz)
-38.4 kS.
-16 bit

Thank you.

andynoack, 2024-01-08:
Direct playback of 16 bit files is problematic as the HackRF only has 8 bit resolution. Generally URH should be able to send signals that can be loaded (also .wav files). Can you describe what happens when you hit the replay button?

Yorch23, 2024-01-08:
Hello, the resolution is not a problem, the problem is found in the bandwidth and the "Sample Rate", when I take captures in sdrsharp with a value greater than 2.0M, everything is ok. If the recording is with a value less than 2.0M, the transmitted carrier is deformed and is no longer valid. URH does not allow you to go below 2.0M to adjust to the recording IQ values, for example it does not allow you to play 1024M. I attach photos of the original signal obtained with AOR DV10 and reproduced directly in sdrsharp perfect and reproduced in hackrf you can see how the carrier widens above 12.5 "Bandwidth".
[image]
[image]
[image]

andynoack, 2024-01-08:
URH does not make sample rates below 2M accessible for the HackRF because they are not within the specification (see https://hackrf.readthedocs.io/en/latest/hackrf_one.html#features).

You could persuade URH by editing the dev/config.py
but this is on your own and will not make it to an URH release of course.

Yorch23, 2024-01-08:
a ok!, thanks for you feedback.

andynoack, 2024-01-08:
You are welcome!

closed 2024-01-08 by andynoack