jopohl/urh #1101: Can not change Device in Device settings because only RTL-TCP is shown

slp14, 2024-02-03:
##### Expected Behavior
I want to use my Pluto-SDR device and therefore it should be possible to change the Device in the device settings of URH from RTL-TCP to Pluto-SDR. I can not find any solution on the web. No screenshot is possible because I use it on my Raspberry-pi 3 (Kali Linux v. 2023.4) and the bug is reported via a different Device.

##### Actual Behavior
The dropdown menu of Device only shows RTL-TCP eventhough I already followed your Readme and checked if the Pluto-SDR is even recognized by the operating system via iio_info - s. There I can find my ADALM-PLUTO SDR and it is also noticable that the mass storage device is seen on the desktop as well. So the device is there. I also installed hackrf-dev and any other dev packages needed.

##### Steps To Reproduce

1. Open terminal and call urh
2. Click on File -> Spectrum Analyzer
3. Try to change device in device settings
4. Only RTL-TCP is shown

##### Platform Specifications

 - OS: Kali Linux
 - URH version: 2.9.6
 - Python version: 3.11.7
 - Installed via pip3

andynoack, 2024-02-04:
Please reinstall via git and make sure the driver is installed beforehand.

slp14, 2024-02-05:
Hello,

when executing python setup.py install it states skipping native support for plutosdr.
Is there any specific package I need to add?
I already added libiio and libad9361 do I need anything else?

All the best

slp14, 2024-02-05:
When forcing the installation with pluto sdr via the command python setup.py --with-plutosdr the plutosdr.cpp can not find iio.h no such file or directory. I did install libiio via the git repository and I can access all the iio commands via my device so I am a bit confused.

slp14, 2024-02-05:
Thank you very much, you are right it is not, I just thought maybe you had a visibility issue regarding other sdrs in your newest release sorry.

closed 2024-02-05 by slp14