jopohl/urh #1102: Recieving Continuous Virus Threat Alerts cont from #1094- Trojan:Win32/Wacatac.B!ml

Hard2Borrow, 2024-02-04:
File Universal.Radio.Hacker-2.9.6-x86.exe still triggers  Virus/ Threat warnings for Trojan: Win32/Wacatac.B!ml

Returns message: This program is dangerous and executes commands from an attacker.

Previous resolution #1094 states to ignore warning/ communicate to anti virus programs that file is not threat.

Please confirm.

##### Screenshots
[image]

jopohl, 2024-02-04:
Yes, please report this to your anti-virus vendor as a false positive. You can see in https://github.com/jopohl/urh/issues/1075 how to report it to Microsoft.

For more context, please refer to [my comment](https://github.com/jopohl/urh/issues/921#issuecomment-992187425) in another related issue.

closed 2024-02-04 by jopohl

Timmmm, 2024-03-02:
FYI both Chrome and Firefox now block this download:

https://github.com/jopohl/urh/releases/download/v2.9.6/Universal.Radio.Hacker-2.9.6-x64.exe

Quite annoying! Honestly it's doubly annoying that they give you no actionable information. "This file is dangerous." No further details. :-/

jopohl, 2024-03-02:
For context, this is caused by PyInstaller, please look here for more information https://github.com/pyinstaller/pyinstaller/issues/6754#issuecomment-1100821249

Possible options:
1. Use a different installation method e.g. with pip
2. Run it from source
3. Report the false-positive to your anti-virus vendor