jopohl/urh #1105: wireless interrogation function with pulse or defined waveform

PatrickDowS, 2024-03-05:
##### Is your feature request related to a problem?

Most of the features are made for RX of Hackrf, but TX with coded or waveform is missing.

##### Describe the solution you'd like

it is good to add a function such as wave1 and wave2 for mixers and modulation.
wave 1 may be sine wave, and wave2 may be square or coded pulse.
frequency can be defined so could make a signal generator and transmitter.

##### Describe alternatives you've considered

it is better if library functions are made such as which can be called by dll(such as windows ). then various features can be expanded by developers.

##### Additional context

PatrickDowS, 2024-03-05:
hope the feature is not combined with gnuradio.

andynoack, 2024-03-06:
You can already add sine waves in Interpretation (please activate the plugin before). Additionally in generation arbitrary data can be generated by choosing the respective frequency properties and inserting binary/hex data.

closed 2024-03-06 by andynoack