jopohl/urh #1110: v2.9.6 sdrplay error

xykts, 2024-03-22:
URH V2.9.6 version cannot load SDRPLAY on Windows 11
It cannot obtain device identifier
Forcing it to start causes an error with the image

[image]
[image]

##### Platform Specifications

 - OS: [WINDOWS 11]
 - URH version: [e.g. 2.9.6]
 - Python version: [e.g. 3.11]
 - Installed via [e.g.Universal.Radio.Hacker-2.9.6-x64.exe ]

andynoack, 2024-03-22:
Can you verify that it worked with URH v2.9.5? Maybe this is a driver related issue.

andynoack, 2024-04-09:
Closed due to inactivity

closed 2024-04-09 by andynoack