jopohl/urh #1113: can not find module distutils

FalcoGer, 2024-04-28:
##### Expected Behavior
urh starts

##### Actual Behavior
```text
Traceback (most recent call last):
  File "/home/user/.local/pipx/venvs/urh/bin/urh", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/user/.local/pipx/venvs/urh/lib/python3.12/site-packages/urh/main.py", line 122, in main
    from urh.controller.MainController import MainController
  File "/home/user/.local/pipx/venvs/urh/lib/python3.12/site-packages/urh/controller/MainController.py", line 24, in <module>
    from urh.controller.dialogs.OptionsDialog import OptionsDialog
  File "/home/user/.local/pipx/venvs/urh/lib/python3.12/site-packages/urh/controller/dialogs/OptionsDialog.py", line 32, in <module>
    from urh.dev.native import ExtensionHelper
  File "/home/user/.local/pipx/venvs/urh/lib/python3.12/site-packages/urh/dev/native/ExtensionHelper.py", line 6, in <module>
    from distutils import ccompiler
ModuleNotFoundError: No module named 'distutils'
```

##### Steps To Reproduce

1. `pipx install urh`
2. `urh`
3. no worky
4. `sudo apt install python3-distutils

```text
sudo apt install python3-distutils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python3-distutils is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python3-distutils' has no installation candidate
```

5. `cd ~/.local/pipx/venvs/urh`
6. `source bin/activate`
7. `python3 -m pip install distutils`

```text
ERROR: Could not find a version that satisfies the requirement distutils (from versions: none)
ERROR: No matching distribution found for distutils
```

##### Screenshots
N/A

##### Platform Specifications
 - OS: Ubuntu MATE (24.04 LTS)
 - URH version: `package urh 2.9.6, installed using Python 3.12.3`
 - Python version: 3.12.3
 - Installed via `pipx`

jopohl, 2024-04-28:
It will work when you replace step 7 with

```
python3 -m pip install setuptools
```

FalcoGer, 2024-04-28:
that did it. but i already had python3-setuptools installed through the package manager. Also that should probably be included in the requirements file or whatever else pipx uses..

closed 2024-04-28 by jopohl

dayo777, 2025-03-23:
> It will work when you replace step 7 with
> 
> ```
> python3 -m pip install setuptools
> ```

When I run the above command, this is the error I get, 
```
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```
I am on Ubuntu 24.04.2 LTS.

Huzzla101, 2025-03-23:
... Python is not Ubuntu its Python, so everyone have been thru that one ,
believe me  =) ---  What you can do is to enter this at the end..

python3 -m pip install setuptools --break-system-packages

Some say its not good in the long run but as long as you dont overdo
it , then it should be ok...
Or you can even make a ext managed env, but that is more complicated tho...

dayo777, 2025-03-24:
> ... Python is not Ubuntu its Python, so everyone have been thru that one ,
> believe me  =) ---  What you can do is to enter this at the end..
>
> python3 -m pip install setuptools [...]

Hi Huzzla, thanks for your explanation. I have been able to resolve this by installing `setuptools` it up in a virtual environment and completed the required tasks I was trying to complete.

Cheers!

Huzzla101, 2025-03-24:
Cheers =)

PR #1114 (2024-04-28): Add setuptools to requirements.txt
Python 3.12 dropped `distutils` from the standard library so we need `setuptools` to run URH. This fixes #1113.