jopohl/urh #1115: Advice needed how to best prepare data for analysis in URH

OevreFlataeker, 2024-05-13:
Hello,

I am doing some experiments with RF with a CC1101 module using an ESP8266 and RTL-SDR.
Using [the SmartRF L](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib) I am able to send an arbitrary message on the ether. I used ASK/OOK modulation, no preamble, default signal strength.

Following the steps in video of this blog post I was able to get back my sent message within seconds:
https://www.rtl-sdr.com/video-tutorial-using-universal-radio-hacker-an-rtl-sdr-and-a-microcontroller-to-clone-433-mhz-remotes/

However I now want to try to use gqrx or any other SDR software to dump the data to a file for later analysis with URH.

I switched gqrx to the exact frequence URH gave me in the spectrum scanner mode, set filter to normal, turned off AGC, cranked up gain to 49 just like in URH and set the demodulator to AM. I can clearly see/hear my signal being sent every 2 seconds in the tool. However, when I export to wav all there is is a single wave ("blib") instead of a signal consisting of encoding my message. 

These pictures show how the signal was perfectly caught in URH. The wav file I have from gqrx just has the very first "bump" and nothing else.

[image]
[image]
[image]
[image]

It seems my settings in gqrx somehow filter the signal to strong for the export. 

I know this sounds more like a problem with me not knowing how to properly use gqrx but I just wanted to understand which settings URH uses when recording so I could maybe reproduce them in gqrx. Other than that are there best practice settings for exporting to WAV and later analysis?

OevreFlataeker, 2024-05-13:
Here are the settings from gqrx. You can clearly see that 433.863420 is not in line with those small dots in the waterfall diagram.

[image]

When I record this and open it in URH I only get

[image]

Cropped to the peak:

[image]

When I manually set the frequency to 433.941.120 which perfectly matches the line of dots and record I get something like this which also doesn't look right?

[image]

Playing with AGC and other settings yields no improvement either. What am I doing wrong?

andynoack, 2024-05-13:
The .wav format is not suitable for saving I/Q data from a SDR, please use a raw I/Q data format like used by GnuRadio or URH.

OevreFlataeker, 2024-05-13:
Oh, but I didn't want to import I/Q, I deliberately wanted to choose WAV. Doesn't URH support WAV according to this? https://github.com/jopohl/urh/wiki/Supported-signal-file-formats

andynoack, 2024-05-13:
URH does support WAV files, but note that the WAV file format does not (!) save clean I/Q data. So if something does not work as expected in URH, do not blame URH but WAV.

OevreFlataeker, 2024-05-13:
I might have a wrong expectation, but I assumed that a signal like shown in the 4th picture from above would actually be what the contents of the WAV file would look like? The thing is: As far as I understood getting information from signals in CTFs or similar tasks is by analyzing the provided WAV files (because they are much smaller than I/Q files), so I wanted to exercise a bit how to properly generate those files so that URH can understand them. As I said initially, I don't intend to blame URH for anything not working, I was just looking for spoilers, that I might be using gqrx in a wrong way? I tried to set the same params like URH seems to use according to the 1st/2nd screenshot.

andynoack, 2024-05-13:
When using WAV files the signals are saved in demodulated form in there. URH demodulates signals itself to allow for playing around with all demodulation settings, so there is a mismatch (Interpretation does not work then).
According to the images the sample rate of your signals might be wrong or mono/stereo does not fit. Maybe there are also some other Wave File format options?

OevreFlataeker, 2024-05-13:
OK, understood, makes perfectly sense. I'll try to read up a little bit about this and learn some more.

andynoack, 2024-05-16:
I'll close here for now. You can also contact us in Slack for further discussions!

closed 2024-05-16 by andynoack