jopohl/urh #1116: SDRplay support in current devices

OevreFlataeker, 2024-05-18:
Working on Win10 with URH 2.96 from the releases section here.

I own both RTL SDR devices and a SDRplay RSPuno. I figured out, that support for those devices needs to be enabled in the Options menu. I can properly find (see the #-number, use them) my RTL-SDR dongles in e.g. the Spectrum Analyzer module, but when I choose SDRplay, the device dropdown stays empty.

Clicking on "Start" gives me the same error as in the issues mentioned below:

SDRPlay: CONNECTED DEVICES: 
SDRPlay-SET DEVICE NUMBER to 0: HARDWARE ERROR (7)

I've read in #783 and #1033 and #1014 but am not sure what I might have been missing? I installed SDRplay API v2 from SDRplay's legacy download section (had v3 from the native installation installed) as I understood v3 is not yet supported - or is it? Could you please clarify if I need gnuradio installed as well? Not sure if I have found the lastest info on this?

If I missed some installation howto how to setup SDRplay with URH, kindly guide me where to find it. Thank you very much.

michelinok, 2025-08-28:
same here....empty dropdown.
did you solve?

stevenjoezhang, 2025-10-18:
I've tried to migrate to the SDRplay v3 API, but I only have an RSP1 and don't have an RSPuno device to test.

Testing of this PR is welcome: #1188