jopohl/urh #1118: URH does not properly interoperate with RTL-SDR Blog v4 stick

OevreFlataeker, 2024-05-31:
I have used the RTL-SDR Blog v3 stick without issues with URH, but now also have a v4 stick which is recognized by URH and there are no errors when recording to showing the spectrum but there is simply no output at all.

The stick is properly detected and shown in the recording options and I pressed a key fob several times. 

[image]

The v4 model needs new drivers as v3 drivers are not compatible with it and I build the new debs for Ubuntu 22.04 following this quickstart guide: https://www.rtl-sdr.com/v4/

```
sudo apt update
sudo apt install libusb-1.0-0-dev git cmake
sudo apt install debhelper

git clone https://github.com/rtlsdrblog/rtl-sdr-blog
cd rtl-sdr-blog
sudo dpkg-buildpackage -b --no-sign
cd ..

sudo dpkg -i librtlsdr0_*
sudo dpkg -i librtlsdr-dev_*
sudo dpkg -i rtl-sdr_*
```

Other tools like gqrx or rtl_433 work (after rebuilding the drivers) w/o issues, but urh does not. I wonder if it is internally still using v3 drivers? I installed urh 2.9.6 using pip.

andynoack, 2024-06-01:
Have you tried to rebuild the cython-Extensions (i.e. drivers)? Go to urh/cythonext/ and run build.py with the current drivers installed. Maybe this fixes your problem.

OevreFlataeker, 2024-06-01:
Doesn´t work. 

```/usr/bin/python3.10: can't open file '/home/daubsi/.local/lib/python3.10/setup.py': [Errno 2] No such file or directory
```
Apparently this file is not present when installing via pip?

```
daubsi@s935:~/.local/lib/python3.10/site-packages/urh/cythonext$ ls -lah .
total 13M
drwxrwxr-x  3 daubsi daubsi 4,0K Mai 31 20:03 .
drwxrwxr-x 15 daubsi daubsi 4,0K Mai 31 20:03 ..
-rw-rw-r--  1 daubsi daubsi  621 Mai 31 20:03 analyze.py
-rwxrwxr-x  1 daubsi daubsi 1,9M Mai 31 20:03 auto_interpretation.cpython-310-x86_64-linux-gnu.so
-rw-rw-r--  1 daubsi daubsi 7,4K Mai 31 20:03 auto_interpretation.pyx
-rwxrwxr-x  1 daubsi daubsi 1,8M Mai 31 20:03 awre_util.cpython-310-x86_64-linux-gnu.so
-rw-rw-r--  1 daubsi daubsi  14K Mai 31 20:03 awre_util.pyx
-rw-rw-r--  1 daubsi daubsi  565 Mai 31 20:03 build.py
-rw-rw-r--  1 daubsi daubsi    0 Mai 31 20:03 __init__.py
-rwxrwxr-x  1 daubsi daubsi 3,0M Mai 31 20:03 path_creator.cpython-310-x86_64-linux-gnu.so
-rw-rw-r--  1 daubsi daubsi 4,4K Mai 31 20:03 path_creator.pyx
drwxrwxr-x  2 daubsi daubsi 4,0K Mai 31 20:03 __pycache__
-rwxrwxr-x  1 daubsi daubsi 3,9M Mai 31 20:03 signal_functions.cpython-310-x86_64-linux-gnu.so
-rw-rw-r--  1 daubsi daubsi  21K Mai 31 20:03 signal_functions.pyx
-rwxrwxr-x  1 daubsi daubsi 2,2M Mai 31 20:03 util.cpython-310-x86_64-linux-gnu.so
-rw-rw-r--  1 daubsi daubsi  250 Mai 31 20:03 util.pxd
-rw-rw-r--  1 daubsi daubsi  14K Mai 31 20:03 util.pyx
```
Do you know if I can enforce the compilation during installation via pip?

andynoack, 2024-06-01:
You can trigger it in the options but it should also be called before the first start of a fresh Installation.

OevreFlataeker, 2024-06-01:
Just tried the manual way:

Purged the URH wheel.
git clone of this repo
python3.10 setup.py bdist_wheel
pip3.10 install .

builds a urh-2.9.6-cp310-cp310-linux_x86_64.whl and installs it

Upon start I ONLY have the RTL-SDR and RTL-NET extension available (rest is gone), but problem persists. 

Where do I see which files are used for this extension? Maybe I need to tweak something so it's forced to look into my RTL-SDR v4 driver source directory during rebuild?

OevreFlataeker, 2024-06-01:
Whoops... just saw that this is explicitly given in the Options dialog :-)

andynoack, 2024-06-01:
Yep, or have a look at the Cython Files in cythonext or dev.
I also saw your first issue but don't have a SDRPlay nor RTLSDR v4 at hand respectively in my lab. So if you figure something out, you are invited to make a Pull Request.

OevreFlataeker, 2024-06-01:
I will give it a try... it's weird.. I see the debs I build during the installation according to the RTL-SDR Blog v4 QSG (see first post) installed 

```
daubsi@s935:/tmp$ dpkg -l | grep librtl
ii  librtlsdr-dev                                               0.7.0~rtlsdrblog1                          amd64        Software defined radio receiver for Realtek RTL2832U (development)
ii  librtlsdr0:amd64                                            0.7.0~rtlsdrblog1                          amd64        Software defined radio receiver for Realtek RTL2832U (library)
```
The version 0.7.0 is the one I compiled yesterday (deb creation date)
But the files shown look older?

```
daubsi@s935:/tmp$ dpkg -L librtlsdr-dev | xargs ls -ld
drwxr-xr-x   21 root root   4096 Okt 13  2021 /.
drwxr-xr-x   17 root root   4096 Okt 12  2023 /usr
drwxr-xr-x  160 root root  24576 Jun  1 18:44 /usr/include
-rw-r--r--    1 root root   1436 Sep 13  2023 /usr/include/rtl-sdr_export.h
-rw-r--r--    1 root root  13622 Sep 13  2023 /usr/include/rtl-sdr.h
drwxr-xr-x  172 root root  12288 Mai 14 23:08 /usr/lib
drwxr-xr-x  154 root root 147456 Jun  1 18:44 /usr/lib/x86_64-linux-gnu
drwxr-xr-x   49 root root   4096 Mai 31 19:21 /usr/lib/x86_64-linux-gnu/cmake
drwxr-xr-x    2 root root   4096 Jun  1 18:44 /usr/lib/x86_64-linux-gnu/cmake/rtlsdr
-rw-r--r--    1 root root    244 Sep 13  2023 /usr/lib/x86_64-linux-gnu/cmake/rtlsdr/rtlsdrConfig.cmake
-rw-r--r--    1 root root   1987 Sep 13  2023 /usr/lib/x86_64-linux-gnu/cmake/rtlsdr/rtlsdrConfigVersion.cmake
-rw-r--r--    1 root root   4179 Sep 13  2023 /usr/lib/x86_64-linux-gnu/cmake/rtlsdr/rtlsdrTargets.cmake
-rw-r--r--    1 root root   1457 Sep 13  2023 /usr/lib/x86_64-linux-gnu/cmake/rtlsdr/rtlsdrTargets-relwithdebinfo.cmake
-rw-r--r--    1 root root  98406 Sep 13  2023 /usr/lib/x86_64-linux-gnu/librtlsdr.a
lrwxrwxrwx    1 root root     14 Sep 13  2023 /usr/lib/x86_64-linux-gnu/librtlsdr.so -> librtlsdr.so.0
drwxr-xr-x    2 root root  12288 Jun  1 18:44 /usr/lib/x86_64-linux-gnu/pkgconfig
-rw-r--r--    1 root root    273 Sep 13  2023 /usr/lib/x86_64-linux-gnu/pkgconfig/librtlsdr.pc
drwxr-xr-x  402 root root  16384 Mai  4 21:47 /usr/share
drwxr-xr-x 3360 root root 131072 Jun  1 10:33 /usr/share/doc
drwxr-xr-x    2 root root   4096 Jun  1 18:44 /usr/share/doc/librtlsdr-dev
-rw-r--r--    1 root root   2803 Sep 13  2023 /usr/share/doc/librtlsdr-dev/changelog.gz
-rw-r--r--    1 root root   7512 Sep 13  2023 /usr/share/doc/librtlsdr-dev/copyright
-rw-r--r--    1 root root   1753 Sep 13  2023 /usr/share/doc/librtlsdr-dev/README.Debian
```
and
```
daubsi@s935:/tmp$ dpkg -L librtlsdr0 | xargs ls -ld
drwxr-xr-x   21 root root   4096 Okt 13  2021 /.
drwxr-xr-x  188 root root  12288 Jun  1 18:44 /etc
drwxr-xr-x    2 root root   4096 Mai 14 23:07 /etc/modprobe.d
lrwxrwxrwx    1 root root      7 Okt 11  2021 /lib -> usr/lib
drwxr-xr-x    4 root root   4096 Mai 12 21:42 /lib/udev
drwxr-xr-x    2 root root  12288 Jun  1 18:44 /lib/udev/rules.d
-rw-r--r--    1 root root   7520 Sep 13  2023 /lib/udev/rules.d/60-librtlsdr0.rules
drwxr-xr-x   17 root root   4096 Okt 12  2023 /usr
drwxr-xr-x  172 root root  12288 Mai 14 23:08 /usr/lib
drwxr-xr-x  154 root root 147456 Jun  1 18:44 /usr/lib/x86_64-linux-gnu
lrwxrwxrwx    1 root root     19 Sep 13  2023 /usr/lib/x86_64-linux-gnu/librtlsdr.so.0 -> librtlsdr.so.0.6git
-rw-r--r--    1 root root  65032 Sep 13  2023 /usr/lib/x86_64-linux-gnu/librtlsdr.so.0.6git
drwxr-xr-x  402 root root  16384 Mai  4 21:47 /usr/share
drwxr-xr-x 3360 root root 131072 Jun  1 10:33 /usr/share/doc
drwxr-xr-x    2 root root   4096 Jun  1 18:44 /usr/share/doc/librtlsdr0
-rw-r--r--    1 root root   2803 Sep 13  2023 /usr/share/doc/librtlsdr0/changelog.gz
-rw-r--r--    1 root root   7512 Sep 13  2023 /usr/share/doc/librtlsdr0/copyright
drwxr-xr-x    2 root root  12288 Jun  1 18:44 /usr/share/metainfo
-rw-r--r--    1 root root   2196 Sep 13  2023 /usr/share/metainfo/librtlsdr0.metainfo.xml
```
and the librtlsdr.so.0.6git in the second output irritates me... I´d assume it should read 0.7.0 somewhere

OevreFlataeker, 2024-06-01:
In fact the compiled so from yesterday is much bigger than the one in /usr/lib/... (though it is indeed still called 0.6)
```
daubsi@s935:/tmp/rtl-sdr-blog$ ls -la obj-x86_64-linux-gnu/src/
total 1872
drwxr-xr-x  3 daubsi daubsi   4096 Mai 31 19:20 .
drwxr-xr-x  5 daubsi daubsi   4096 Mai 31 19:20 ..
drwxr-xr-x 13 daubsi daubsi   4096 Mai 31 19:20 CMakeFiles
-rw-r--r--  1 daubsi daubsi  11197 Mai 31 19:20 cmake_install.cmake
-rw-r--r--  1 daubsi daubsi  83482 Mai 31 19:20 libconvenience_static.a
-rw-r--r--  1 daubsi daubsi 906526 Mai 31 19:20 librtlsdr.a
lrwxrwxrwx  1 daubsi daubsi     14 Mai 31 19:20 librtlsdr.so -> librtlsdr.so.0
lrwxrwxrwx  1 daubsi daubsi     19 Mai 31 19:20 librtlsdr.so.0 -> librtlsdr.so.0.6git
-rwxr-xr-x  1 daubsi daubsi 334128 Mai 31 19:20 librtlsdr.so.0.6git
-rw-r--r--  1 daubsi daubsi  29652 Mai 31 19:20 Makefile
-rwxr-xr-x  1 daubsi daubsi  60184 Mai 31 19:20 rtl_adsb
-rwxr-xr-x  1 daubsi daubsi  30440 Mai 31 19:20 rtl_biast
-rwxr-xr-x  1 daubsi daubsi  44552 Mai 31 19:20 rtl_eeprom
-rwxr-xr-x  1 daubsi daubsi  97280 Mai 31 19:20 rtl_fm
-rwxr-xr-x  1 daubsi daubsi  87952 Mai 31 19:20 rtl_power
-rwxr-xr-x  1 daubsi daubsi  49600 Mai 31 19:20 rtl_sdr
-rw-r--r--  1 daubsi daubsi   1007 Mai 31 19:20 rtlsdr_export.h
-rw-r--r--  1 daubsi daubsi   1062 Mai 31 19:20 rtlsdr_static_export.h
-rwxr-xr-x  1 daubsi daubsi  83704 Mai 31 19:20 rtl_tcp
-rwxr-xr-x  1 daubsi daubsi  52336 Mai 31 19:20 rtl_test
daubsi@s935:/tmp/rtl-sdr-blog$ dpkg -L librtlsdr0 | xargs ls -ld | grep git
lrwxrwxrwx    1 root root     19 Sep 13  2023 /usr/lib/x86_64-linux-gnu/librtlsdr.so.0 -> librtlsdr.so.0.6git
-rw-r--r--    1 root root  65032 Sep 13  2023 /usr/lib/x86_64-linux-gnu/librtlsdr.so.0.6git
```
I'll trace the creation of the debs according to the RTL-SDR Blog v4 QSG. Maybe something went wrong here...

OevreFlataeker, 2024-06-01:
OK, got it working.

This is the step by step guide how to properly set it up:

Follow instructions at https://www.rtl-sdr.com/v4/, this means:

```
sudo apt update
sudo apt install libusb-1.0-0-dev git cmake
sudo apt install debhelper
cd /tmp
git clone https://github.com/rtlsdrblog/rtl-sdr-blog
cd rtl-sdr-blog
sudo dpkg-buildpackage -b --no-sign
cd ..

sudo dpkg -i librtlsdr0_*.deb
sudo dpkg -i librtlsdr-dev_*.deb
sudo dpkg -i rtl-sdr_*.deb
```
(Note that we cloned to /tmp/rtl-sdr-blog and built there)

Check that "dpkg -l | grep rtlsdrblog" lists the 3 new packages:

```
daubsi@s935:/tmp$ dpkg -l | grep rtlsdrblog
ii  librtlsdr-dev                                               0.7.0~rtlsdrblog1                          amd64        Software defined radio receiver for Realtek RTL2832U (development)
ii  librtlsdr0:amd64                                            0.7.0~rtlsdrblog1                          amd64        Software defined radio receiver for Realtek RTL2832U (library)
ii  rtl-sdr                                                     0.7.0~rtlsdrblog1                          amd64        Software defined radio receiver for Realtek RTL2832U (tools)
```

Install urh via pip:

```
pip3 install urh
```

Start urh and go to Edit->Options

In the textbox labeled "Library directories" add: 
```/tmp/rtl-sdr-blog/obj-x86_64-linux-gnu/src```
In the textbox labeled "Include directories" add: 
```/tmp/rtl-sdr-blog/include```

Click "rebuild extensions". 
Now the RTL-SDR extension is rebuild using the V4 files (actually ONLY this one is build as all the dev files for the other extensions are not present on my system)

Restart URH

Go to record signal, record at 433 MHz and send some signals (in my case, some demo signals from a Flipperzero). You should immediately see some signals being recorded.

andynoack, 2024-06-01:
Great stuff, thank you very much!

OevreFlataeker, 2024-06-01:
My pleasure. Thanks for the great tool! I've just gone through the full complete uninstall, install, rebuild cycle once more and verified it only worked after the manual rebuild -> guide works as described.

closed 2024-06-04 by andynoack