jopohl/urh #1119: urh 2.6.9 compiled with USRP support - but USRP isn't appearing in the Devices dropdowns?

victorhooi, 2024-06-06:
##### Expected Behavior

Installing urh with USRP support, it should list "USRP" as an option under Devices in the UI.

(I saw that USRP support was added quite some time back in 2017 - https://github.com/jopohl/urh/issues/133 - hence why I assumed it should probably work out of the box).

##### Actual Behavior

I've installed URP 2.9.6 with USRP support, and I don't seem to see "USRP" as an option in the Device dropdowns.

If I launch urh, and then go to any of the dialogs (e.g. Spectrum Analyzer), and then click on Device - I don't seem to see "USRP" (or anything similar) listed:

[image]

##### Steps To Reproduce

1. Install urh via nixpkgs ([package link](https://search.nixos.org/packages?channel=24.05&show=urh&from=0&size=50&sort=relevance&type=packages&query=urh), [package source](https://github.com/NixOS/nixpkgs/blob/master/pkgs/applications/radio/urh/default.nix))
2. From a terminal, run "urh".
3. Go to File, then Spectrum Analyzer, then click on the "Device" dropdown.
4. Look at the list of devices.

##### Screenshots

##### Platform Specifications

 - OS: macOS 14.4.1
 - URH version: 2.6.9
 - Python version: [e.g. 3.7]
 - Installed via [Nix packager](https://search.nixos.org/packages?channel=24.05&show=urh&from=0&size=50&sort=relevance&type=packages&query=urh)

I also checked my local build logs for the nix package - it does seem to mention it was compiled with USRP support.

```
Found hackrf lib. Will compile with native hackrf support
Found LimeSuite lib. Will compile with native limesdr support
Found iio lib. Will compile with native plutosdr support
Found rtlsdr lib. Will compile with native rtlsdr support
Found uhd lib. Will compile with native usrp support
Skipping native support for sdrplay
[ 1/12] Cythonizing src/urh/cythonext/auto_interpretation.pyx
[ 2/12] Cythonizing src/urh/cythonext/awre_util.pyx
[ 3/12] Cythonizing src/urh/cythonext/path_creator.pyx
[ 4/12] Cythonizing src/urh/cythonext/signal_functions.pyx
```
Full build logs are here:

https://gist.github.com/victorhooi/540a01284a681cb33cb5aea32fa10ff1

victorhooi, 2024-06-06:
I went to https://github.com/jopohl/urh/wiki/Supported-devices, and I saw that it mentioned there's an "Options -> Devices" screen.

On macOS, this seems to be under python3.11, Preferences...:

[image]

If I open that up, and go to the Devices tab, I noticed that for some reason, USRP was unticked by default:

[image]

I'm thinking that might be my issue - will confirm once I get home.

However, is there a reason that USRP is unticked by defaut? As in, are there any caveats or warnings with this plugin versus the other types?

andynoack, 2024-06-07:
Usually all SDRs are unticked when you install from source. As you have installed with a package manager, there might be a special crafted package with enabled devices. The driver, however, must be installed beforehand, so in your case the USRP was probably disabled at first. If the driver is installed you should be able to activate it now.

Source975, 2024-08-06:
airspy -- ERROR: No module named 'urh.dev.native.lib.airspy'
bladerf -- ERROR: No module named 'urh.dev.native.lib.bladerf'
hackrf -- ERROR: No module named 'urh.dev.native.lib.hackrf'
limesdr -- ERROR: No module named 'urh.dev.native.lib.limesdr'
plutosdr -- ERROR: No module named 'urh.dev.native.lib.plutosdr'
rtlsdr -- ERROR: No module named 'urh.dev.native.lib.rtlsdr'
sdrplay -- ERROR: No module named 'urh.dev.native.lib.sdrplay'
usrp -- ERROR: No module named 'urh.dev.native.lib.usrp'
SoundCard -- No module named 'pyaudio'

andynoack, 2024-08-06:
I will close here because the original issue seems to be solved or inactive. Issues are not for installation support, please visit our Slack channel.

closed 2024-08-06 by andynoack

andynoack, 2024-08-06:
This is not a bug in URH, your installation (Windows/SDR drivers/URH installation) is bugged. Make a clean reinstallation.

mono7s, 2025-08-01:
> airspy -- ERROR: No module named 'urh.dev.native.lib.airspy' bladerf -- ERROR: No module named 'urh.dev.native.lib.bladerf' hackrf -- ERROR: No module named 'urh.dev.native.lib.hackrf' limesdr -- [...]

Hello, may I ask if your issue has been resolved? I am facing the same problem and I need your help.

mono7s, 2025-08-01:
> airspy -- ERROR: No module named 'urh.dev.native.lib.airspy' bladerf -- ERROR: No module named 'urh.dev.native.lib.bladerf' hackrf -- ERROR: No module named 'urh.dev.native.lib.hackrf' limesdr -- [...]

Or could you please tell me how to successfully use Limesdr after compiling the source code on the Windows platform?