jopohl/urh #1121: Recorded data with rtl_tcp seems weird

OevreFlataeker, 2024-06-07:
I have rtl_tcp running on a remote system, connected to an RTL SDR Blog v3 stick.
Using rtl_tcp with tools like gqrx or rtl_433 with the source defined remotely (rtl_tcp) works fine.
Using urh, I can use the spectrum analyzer and also this looks fine.
But actually recording doesn´t work properly.
It looks as if the signal is glued to the top of the window and has long vertical lines. I can see impulses when I send something (e.g. click a key fob), but the recorded data is not parseable either. 
There is no error otherwise.

Display when nothing is detected
[image].

Pressing a key/sending a signal
[image]

In addition the address of the remote rtl_tcp is reset on every usage to 127.0.0.1. The previous value is not saved. Also using a DNS name would be nice here

Some sample recordings
[signals.zip](https://github.com/user-attachments/files/15743975/signals.zip)

URH 2.9.6 on Ubuntu 22.04. rtl_tcp from Ubuntu package rtl-sdr, Version 0.6.0-4

andynoack, 2024-06-08:
This looks like a format mismatch somehow, rtl_tcp outputs a .complex16u format. The attached signals, however, seem to be in a wrong format, there does not seem to be any information in it (most information is very near to -1.0 or 1.0).

OevreFlataeker, 2024-06-08:
Well, I agree it look "mismatched" - the question is, what I can do about it? The only option I see which might actually have an effect is the sample rate itself but I don't see anything about the format?

```
rtl_tcp, an I/Q spectrum server for RTL2832 based DVB-T receivers

Usage:  [-a listen address]
        [-p listen port (default: 1234)]
        [-f frequency to tune to [Hz]]
        [-g gain (default: 0 for auto)]
        [-s samplerate in Hz (default: 2048000 Hz)]
        [-b number of buffers (default: 15, set by library)]
        [-n max number of linked list buffers to keep (default: 500)]
        [-d device index (default: 0)]
        [-P ppm_error (default: 0)]
        [-T enable bias-T on GPIO PIN 0 (works for rtl-sdr.com v3 dongles)]
```

andynoack, 2024-06-08:
Hmm, difficult to suggest a solution here! Apparently there are different versions of rtl_tcp and this version either doesn't record data in a clean way or is not compatible with the standard socket implementation used in URH. Can you have a look at the network trace to find out, if it is a problem on the network or SDR interface side?

OevreFlataeker, 2024-06-08:
But wouldn't other tools like rtl_433 and gqrx then also have a problem with a wrong transmitted data type? Or are they that forgiving?

I think I'll first try to build rtl_tcp from sources and check if that makes any difference.

Can I verify your assumption it's the wrong datatype with tcpdump/wireshark?

andynoack, 2024-06-08:
Well, the RTLSDR is not a real SDR but a misused DVB-T stick, therefore precision and so on is not as good as in "normal" SDRs, see e.g. the castrated I/Q format (.complex16u). There are tools optimized for this special RTLSDR (like those you mentioned) but URH is rather designed for real SDRs like the USRP or HackRF. I can image that those tools are more forgiving in consequence and maybe also have support for the latest RTLSDR hardware revisions (which may be not the case for default URH). I know, however, that RTLSDRs and also the TCP variant have been working for many years, because it was among others my part to implement years ago.
If you can observe the TCP stream in wireshark (I would prefer it over tcpdump here), you should predominantly see 16 bit (4 nibble) packages that consist of values near 00 or ff (i.e. -1.0 or 1.0 I/Q) if something with the data does not really work out.

mplough, 2025-05-29:
I've seen what appears to be the exact same issue except with an RTL-SDR v3 plugged into my local machine.

I tested with urh version 2.9.8 from Homebrew on an M3 Mac and rtl_433 version string as follows;

```
rtl_433 version 25.02 (2025-02-19) inputs file rtl_tcp RTL-SDR with TLS
```

## Steps to reproduce

First, I saved a file with `rtl_433` while pressing buttons on a remote control:
```
rtl_433 -f 315000000 -s 2000000 -g 20 -W remote3.complex16u
```
I then load that file into `urh` and it shows up properly.  Notice how we have some blips over a center line.

[image]

I then started up `rtl_tcp` and saved a file in urh.  Settings follow.  Note the line at the top of the signal window with a bunch of stuff blipping down from it:

[image]

And of course, attempting to interpret this incorrectly recorded signal does not work:

[image]

## Interpretation
This does not appear to be an issue with the limited precision capabilities of an RTL-SDR; instead it's a data issue somewhere in the chain.  I suspect it's either an issue with interpreting what's coming back from `rtl_tcp` during the recording phase as the interpretation phase works fine with pre-recorded data from the same device recorded only moments before on the same machine.

PR #1167 (2025-06-01): Fix receiving data with rtl_tcp
This PR fixes #1121.

I have tested this with `rtl_tcp` and an RTL-SDR v3 dongle, plus a remote that sends unknown data on 315 MHz.

Both the recording and spectrogram look good; see screenshots.

[image]
[image]

Data recorded with RTL-TCP mode is now saved as `.complex16s` rather than `.complex16u` and the data can now be used for analysis.
closed 2025-06-01 by andynoack