jopohl/urh #1123: urh error

ICantMakeThings, 2024-06-22:
##### Expected Behavior
to work

##### Actual Behavior

Traceback (most recent call last):
  File "/home/nulli/.local/bin/urh", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/nulli/.local/share/pipx/venvs/urh/lib/python3.12/site-packages/urh/main.py", line 105, in main
    import urh.cythonext.signal_functions
  File "src/urh/cythonext/signal_functions.pyx", line 1, in init urh.cythonext.signal_functions
ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject

reinstalling numpy does not work

##### Steps To Reproduce

pipx install urh
pipx run urh
error.

##### Platform Specifications

 - OS: ubuntu 24.04
 - URH version: newest
 - Python version: 3.12.3
 - Installed via pipx (pip gives some error i cant fix, numpy installed with pipx too)

LoicPZ, 2024-06-27:
try this 
`pipx runpip urh install numpy==1.26.4`
`pipx runpip urh install setuptools`

andrew-stclair, 2024-07-03:
That worked for me, though setuptools was already installed.

TL;DR just needed to install numpy==1.26.4 instead of numpy==2.0.0

godmar, 2024-07-26:
Without setuptools, on recent versions of Python, you get
```
ModuleNotFoundError: No module named 'distutils'
```

I believe the numpy dependency could be specify in [`requirements.txt`](https://github.com/jopohl/urh/blob/544efd35ac4e0105cb63a31f2dc209c3834bc7bd/data/requirements.txt) perhaps.

closed 2024-11-11 by jopohl

jopohl, 2024-11-11:
Thanks for reporting. Fixed with [v2.9.8](https://github.com/jopohl/urh/releases/tag/v2.9.8), now a regular `pipx install urh` works.