jopohl/urh #1125: Demodulated .wav gets converted to Signal after restart

Peugot206, 2024-06-28:
##### Expected Behavior
When I import a .wav file it becomes 'Demodulated (1-channel) .wav'. When I then close the application I want it to stay like this. Or when I simply remove and add the signal again I want it to stay like that.

##### Actual Behavior
Whenever I import a .wav file it gets important as following:
[image]

Then if I close this signal by pressing the X button (not from the application, but the one in the image) and then re-open it, OR if I simply restart the application it gets converted to a 'Signal', as following:
[image]

Which results in completely different data sadly. That is also one of the reasons why I import .wav files, as the recording function does not seem to work for my signals. The data is completely different than the .wav file captured in SDR Sharp.

##### Steps To Reproduce

1. Open a .wav file
2. Close/remove the .wav file
3. Re-open the .wav file again, or simply re-start the application
4. The 'Interpretation' of the signals now changed, which changes the signals and data for me.

##### Screenshots
See above.

##### Platform Specifications

 - OS: Windows 10
 - URH version: 2.9.6
 - Python version: Not sure..
 - Installed via Windows installer

Thanks for this amazing software!

andynoack, 2024-06-28:
.wav is not capable of saving the required information for SDR recordings, therefore please use the standard .complex format. When you are importing .wav files URH fakes the missing data once.

Peugot206, 2024-07-01:
Thanks for the reply. I didn't know that.

The reason I'm doing it this way is because when I record the signal using URH the results are incorrect. The waveform does not look close to the signal I'm sending. It looks like URH adds its own layer of modulation/filtering already on the signal (that I don't seem to be able to disable). It's then also not able to extract the data I'm looking for.

However when using the exact same PC, RTL-SDR, and record the signal in Airspy SDR# and import that .wav file into URH I am able to get the correct data from it. The record settings are (as far as they can be configured) the same in SDR# and URH. 

Any clue why that is? Am I doing something wrong here? If I can record the signal directly in URH the right way, that'd be perfect.

andynoack, 2024-07-01:
I can tell that URH records the plain data that are coming from the SDR, it is completely vanilla. Maybe SDR# does some type of filtering, whereby I can image that you can finetune or postprocess the URH recording (with URH) too. Have you tried the filters in interpretation?

Peugot206, 2024-07-02:
Interesting. Here are my SDR# settings (controller and audio recorder): 
[image]

And here are my URH settings (I already tried different sample rates):
[image]

The one with SDR# works and are as expected. The signal recorded with URH looks very odd.

andynoack, 2024-07-02:
Actually, these settings do not make sense. The frequency is not on the spot, bandwidth differs from sample rate (while sample rate is also an odd value) and frequency correction seems arbitrary. It seems to me that this is not an URH issue but a configuration thing, so I suggest to continue on Slack where we can discuss/help in a dialog. I'll close here for now.

closed 2024-07-02 by andynoack

Peugot206, 2024-07-05:
Thanks for the reply. Not sure why they don't make sense. It's a very specific frequency this device uses. I also don't see why the bandwidth should be the same as the sample rate? 

It would really help if there was a setting to disable this automatic conversion and if the cutted out parts gets saved as well (this does not seem to be the case right now). This program does then exactly what we need perfectly :).

andynoack, 2024-07-05:
There are absolutely no "automatic conversions" or something like that! As well, no parts are cutted out, URH records exactly what is coming from the SDR.

Your assumptions are misleading for people reading through our issues and that is why, I do not want such things here. Maybe SDR# works better for your use case because they do some magic postprocessing or automatic filtering. We don't.

Peugot206, 2024-07-05:
Thanks for the reply. There is a conversion... one way or another.
From this
[image]

to this:
[image]

After which the data is incorrect, and URH treats my data different. While in the first screenshot it is correct.

> As well, no parts are cutted out

I'm talking about my own cuts.. I make cuts/trim in the first image, but after a restart these cuts/trim are lost as well.

Don't get me wrong, I absolutely love this program! It's just frustrating that after a restart it's all lost. So sadly no, SDR# is not better suited for me.

andynoack, 2024-07-05:
There must be a conversion, right. This is, however, related to the file format and not the recording.
You can avoid this by using the .complex file format.