jopohl/urh #1128: IQ csv file demodulate

kiran809576, 2024-07-16:
Hi,

I have a csv file with IQ samples. Instead of doing demodulation using the exe in windows. I want to simply give this file as argument in a Linux system and store the demodulated binary data in a text file.
Simply, instead of using application in windows, i want to run like commands in Linux. 
Someone please help me out.

Thank you.

andynoack, 2024-07-16:
Hey, please have a look in our Wiki: https://github.com/jopohl/urh/wiki/Command-Line-Interface

closed 2024-07-16 by andynoack

kiran809576, 2024-07-16:
If I just give the IQ file as input, will this command line interface autodetect the parameters? Or do I need to pass the required parameters for demodulation as the input arguments?