jopohl/urh #1130: On windows 11 platform, using run from source.usrp and other related devices are not available

Source975, 2024-08-07:
On windows 11 platform, using run from source.
1) In a conda environment with python version 3.9.0, the run reports the following error: 1.
Traceback (most recent call last):
  File "d:\communication_signal_detection\urh-master\urh-master\src\urh\main.py", line 9, in <module>
    Importing QTimer, Qt from PyQt5.QtCore
ImportError: DLL load failed while importing QtCore: The specified module could not be found.
2) In a native environment with python version 3.11.8, the programme runs, but usrp and other related devices are not available,along with a large number of warnings: 
Could not find C++ extensions, trying to build them.
open.c
LINK : fatal error LNK1181: 无法打开输入文件“airspy.lib”
Skipping native support for airspy
bladerf_open.c
LINK : fatal error LNK1181: 无法打开输入文件“bladeRF.lib”
Skipping native support for bladerf
hackrf_init.c
LINK : fatal error LNK1181: 无法打开输入文件“hackrf.lib”
Skipping native support for hackrf
LMS_GetDeviceList.c
LINK : fatal error LNK1181: 无法打开输入文件“LimeSuite.lib”
Skipping native support for limesdr
iio_create_default_context.c
LINK : fatal error LNK1181: 无法打开输入文件“iio.lib”
Skipping native support for plutosdr
rtlsdr_get_device_name.c
LINK : fatal error LNK1181: 无法打开输入文件“rtlsdr.lib”
Skipping native support for rtlsdr
uhd_usrp_find.c
LINK : fatal error LNK1181: 无法打开输入文件“uhd.lib”
Skipping native support for usrp
mir_sdr_ApiVersion.c
LINK : fatal error LNK1181: 无法打开输入文件“mir_sdr_api.lib”
Skipping native support for [sdrplay](url)
[error.txt](https://github.com/user-attachments/files/16517447/error.txt)

Source975, 2024-08-07:
[image]

Source975, 2024-08-07:
[image]

Source975, 2024-08-07:
64-bit operating system, x64-based processor

andynoack, 2024-08-07:
Clearly, your build environment is missing some packages (e.g. QtCore + SDR drivers)! Issues are NOT for installation support.

closed 2024-08-07 by andynoack

Source975, 2024-08-07:
> Clearly, your build environment is missing some packages (e.g. QtCore + SDR drivers)! Issues are NOT for installation support.

So what am I supposed to do about it?

Source975, 2024-08-07:
airspy -- ERROR: No module named 'urh.dev.native.lib.airspy'
bladerf -- ERROR: No module named 'urh.dev.native.lib.bladerf'
hackrf -- ERROR: No module named 'urh.dev.native.lib.hackrf'
limesdr -- ERROR: No module named 'urh.dev.native.lib.limesdr'
plutosdr -- ERROR: No module named 'urh.dev.native.lib.plutosdr'
rtlsdr -- ERROR: No module named 'urh.dev.native.lib.rtlsdr'
sdrplay -- ERROR: No module named 'urh.dev.native.lib.sdrplay'
usrp -- ERROR: No module named 'urh.dev.native.lib.usrp'
SoundCard -- No module named 'pyaudio'