jopohl/urh #1132: Getting no signals pls help

Austriana, 2024-08-18:
Hey,
I am using an RTL SDR V4 on Win 10.
My problem is i dont get any Signals in. 
When i go to SDR sharp everything works fine.
Anyone experience the same issue?

Marco

Austriana, 2024-08-18:
[image]

Austriana, 2024-08-18:
[image]

andynoack, 2024-08-20:
Looks like a duplicate of #1118. Does the solution included in the mentioned issue work for you?

Austriana, 2024-08-20:
Thanks for the tipp! I am very busy at work rn. But i will try that as fast as possible!

Austriana, 2024-08-22:
Hey @andynoack !!
It works, i got the fix with your Help!! Thank you very much :)

Marco

Austriana, 2024-08-22:
If anyone having the same issue follow this link https://github.com/jopohl/urh/issues/1132#issuecomment-2299916012

closed 2024-08-22 by Austriana

Austriana, 2024-08-22:
If anyone have the same issue ---> https://www.rtl-sdr.com/v4/