jopohl/urh #1133: LINK : fatal error LNK1181: 无法打开输入文件“airspy.lib”

Source975, 2024-08-18:
##### Expected Behavior

I want to run the programme from the source code, the first time I run the main programme, it prompts the following error. devices such as USRP cannot be connected for use, what should I do to fix it?

##### Actual Behavior
 C:/Users/71937/.conda/envs/usrp/python.exe d:/communication_signal_detection/urh-master/urh-master/src/urh/main.py
##### Steps To Reproduce

PS D:\communication_signal_detection\urh-master> & C:/Users/71937/.conda/envs/usrp/python.exe d:/communication_signal_detection/urh-master/urh-master/src/urh/main.py
Could not find C++ extensions, trying to build them.
open.c
LINK : fatal error LNK1181: 无法打开输入文件“airspy.lib”
Skipping native support for airspy
bladerf_open.c
LINK : fatal error LNK1181: 无法打开输入文件“bladeRF.lib”
Skipping native support for bladerf
hackrf_init.c
LINK : fatal error LNK1181: 无法打开输入文件“hackrf.lib”
Skipping native support for hackrf
LMS_GetDeviceList.c
LINK : fatal error LNK1181: 无法打开输入文件“LimeSuite.lib”
Skipping native support for limesdr
iio_create_default_context.c
LINK : fatal error LNK1181: 无法打开输入文件“iio.lib”
Skipping native support for plutosdr
rtlsdr_get_device_name.c
LINK : fatal error LNK1181: 无法打开输入文件“rtlsdr.lib”
Skipping native support for rtlsdr
uhd_usrp_find.c
LINK : fatal error LNK1181: 无法打开输入文件“uhd.lib”
Skipping native support for usrp
mir_sdr_ApiVersion.c
LINK : fatal error LNK1181: 无法打开输入文件“mir_sdr_api.lib”
Skipping native support for sdrplay
running build_ext
Finalizing options
building 'urh.cythonext.auto_interpretation' extension
building 'urh.cythonext.awre_util' extension
building 'urh.cythonext.path_creator' extension
building 'urh.cythonext.signal_functions' extension
building 'urh.cythonext.util' extension

##### Screenshots

##### Platform Specifications

 - OS:windows 11 /23H2
 - URH version: 2.9.6
 - Python version: 3.9
 - Installed via pip

Source975, 2024-08-18:
[image]

Source975, 2024-08-19:
When the following is entered:
python setup.py --with-airspy --with-bladerf --with-hackrf --with-limesdr --with-plutosdr --with-rtlsdr  --with-sdrplay --with-usrp  install
The error is reported as follows:
(usrp) D:\communication_signal_detection\urh-master\urh-master>python setup.py --with-airspy --with-bladerf --with-hackrf --with-limesdr --with-plutosdr --with-rtlsdr  --with-sdrplay --with-usrp  install
get_package_data
*******************************************222222222222222***************************************************
*********************************************111111111111*************************************************
********************************************3333333333333**************************************************
['auto_interpretation', 'awre_util', 'path_creator', 'signal_functions', 'util']
********************************************3333333333334**************************************************
********************************************3333333333335**************************************************
********************************************1**************************************************
********************************************2**************************************************
********************************************3**************************************************
********************************************4**************************************************
********************************************5**************************************************
********************************************6**************************************************
********************************************7**************************************************
********************************************8**************************************************
********************************************81**************************************************
Enforcing native airspy support
airspy
{}
<setuptools.extension.Extension('urh.dev.native.lib.airspy') at 0x24cc3af1760>
********************************************81**************************************************
Enforcing native bladerf support
bladerf
{}
    Environment variable BLADERF_API_VERSION is unset, try to automatically detect API version
get_api_version.c
C:\Users\71937\AppData\Local\Temp\urh-z21vvajf\get_api_version.c(3): fatal error C1083: 无法打开包括文件: “libbladeRF.h”: No such file or directory
    API version check failed: command 'D:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.40.33807\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
    Using BLADERF_API_VERSION=0.0
<setuptools.extension.Extension('urh.dev.native.lib.bladerf') at 0x24cc3af1790>
********************************************81**************************************************
Enforcing native hackrf support
hackrf
C:\Users\71937\AppData\Local\Temp\urh-65vtfk3r\hackrf_open_by_serial.c
********************************************0**************************************************
hackrf_open_by_serial.c
LINK : fatal error LNK1181: 无法打开输入文件“hackrf.lib”
********************************************false**************************************************
Skipping HACKRF_MULTI_DEVICE_SUPPORT as installed driver does not support it
{'BLADERF_API_VERSION': 0.0, 'HACKRF_MULTI_DEVICE_SUPPORT': 0}
<setuptools.extension.Extension('urh.dev.native.lib.hackrf') at 0x24cc3af1820>
********************************************81**************************************************
Enforcing native limesdr support
limesdr
{'BLADERF_API_VERSION': 0.0, 'HACKRF_MULTI_DEVICE_SUPPORT': 0}
<setuptools.extension.Extension('urh.dev.native.lib.limesdr') at 0x24cc6408670>
********************************************81**************************************************
Enforcing native plutosdr support
plutosdr
{'BLADERF_API_VERSION': 0.0, 'HACKRF_MULTI_DEVICE_SUPPORT': 0}
<setuptools.extension.Extension('urh.dev.native.lib.plutosdr') at 0x24cc64084f0>
********************************************81**************************************************
Enforcing native rtlsdr support
rtlsdr
C:\Users\71937\AppData\Local\Temp\urh-tbdtcxq3\rtlsdr_set_tuner_bandwidth.c
********************************************0**************************************************
rtlsdr_set_tuner_bandwidth.c
LINK : fatal error LNK1181: 无法打开输入文件“rtlsdr.lib”
********************************************false**************************************************
Skipping RTLSDR_BANDWIDTH_SUPPORT as installed driver does not support it
{'BLADERF_API_VERSION': 0.0, 'HACKRF_MULTI_DEVICE_SUPPORT': 0, 'RTLSDR_BANDWIDTH_SUPPORT': 0}
<setuptools.extension.Extension('urh.dev.native.lib.rtlsdr') at 0x24cc6408460>
********************************************81**************************************************
Enforcing native usrp support
usrp
{'BLADERF_API_VERSION': 0.0, 'HACKRF_MULTI_DEVICE_SUPPORT': 0, 'RTLSDR_BANDWIDTH_SUPPORT': 0}
<setuptools.extension.Extension('urh.dev.native.lib.usrp') at 0x24cc6408640>
********************************************81**************************************************
Enforcing native sdrplay support
sdrplay
{'BLADERF_API_VERSION': 0.0, 'HACKRF_MULTI_DEVICE_SUPPORT': 0, 'RTLSDR_BANDWIDTH_SUPPORT': 0}
<setuptools.extension.Extension('urh.dev.native.lib.sdrplay') at 0x24cc6408430>
[... cut ...]
********************************************3333333333337**************************************************
********************************************44444444444444444**************************************************
running install
running bdist_egg
running egg_info
writing src\urh.egg-info\PKG-INFO
writing dependency_links to src\urh.egg-info\dependency_links.txt
writing entry points to src\urh.egg-info\entry_points.txt
writing requirements to src\urh.egg-info\requires.txt
writing top-level names to src\urh.egg-info\top_level.txt

Source975, 2024-08-19:
[image]
Is there a problem with this logic?

andynoack, 2024-08-20:
No, there isn't a problem with URH. But there is a Problem with your installation. Dude, again (!!!), fix your installation and install drivers for the SDRs you want to use.

closed 2024-08-20 by andynoack

Source975, 2024-08-21:
How should I install drivers for USRP.

Source975, 2024-08-21:
Won't the urh software build install the sdr driver when running from source?