jopohl/urh #1144: Unable to run pipx version: "ValueError: numpy.dtype size changed, may indicate binary incompatibility."

zt64, 2024-11-09:
##### Expected Behavior

urh should launch the GUI.

##### Actual Behavior
```
Traceback (most recent call last):
  File "/home/psi/.local/bin/urh", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/psi/.local/share/pipx/venvs/urh/lib/python3.12/site-packages/urh/main.py", line 105, in main
    import urh.cythonext.signal_functions
  File "src/urh/cythonext/signal_functions.pyx", line 1, in init urh.cythonext.signal_functions
ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject
```

##### Steps To Reproduce

1. Install urh with `pipx install urh`
2. Run `urh` in the terminal
3. Error occurs

##### Platform Specifications

 - OS: Arch Linux
 - URH version: 2.9.6
 - Python version: 3.12.7
 - Installed via pipx

andynoack, 2024-11-09:
Hey, this seems to be a common issue. Downgrading the numpy version seems to be the way to go, see e.g. https://github.com/Sinaptik-AI/pandas-ai/issues/1251

denizea, 2024-11-10:
Following helped me to install: `pipx install urh --pip-args="numpy==1.26.4"`

PR #1145 (2024-11-11): Requirements: Pin to numpy 1.x.x
As suggested [here](https://github.com/jopohl/urh/issues/1123#issuecomment-2252712620) pin the numpy version to prevent errors with pipx.

This fixes #1144 and fixes #1122.
closed 2024-11-11 by jopohl