jopohl/urh #1149: Please port to numpy-2

yarda, 2025-01-06:
##### Is your feature request related to a problem?

Please port to numpy-2. Some distros (e.g. Fedora) no more provide numpy<2.

##### Describe the solution you'd like

The GUI works with numpy-2.

##### Describe alternatives you've considered

No way to distro package it to fedora-42, because there is no numpy-1

##### Additional context

numpy-2 migration guide: https://numpy.org/devdocs/numpy_2_0_migration_guide.html
Maybe I could port it myself, but it will take time, so at least reporting it for the RFE to be tracked.

yarda, 2025-01-06:
BTW I compiled it with numpy-2.2.1 and it seems it runs fine, at least I didn't observe the startup crash. Maybe if compiled with numpy-2 it can be run with numpy-2? And the crash will occur only if compiled with numpy-2 and run with numpy-1? If this is the case, the numpy dep should be relaxed.

andynoack, 2025-01-06:
Hey @yarda,
thanks for trying out! We appreciate a pull request if you are able to get it running with the current numpy version. The unit tests of URH should inidicate whether there are problems.
Best regards!

yarda, 2025-01-06:
Sure, I will check and I will also enable the unit tests during Fedora build process. Could you point me to instructions how to run them all? I tried `python3 setup.py test` but it seems this is not the correct way.

andynoack, 2025-01-06:
You can execute the tests manually or with PyCharm for example. You can, however, also use the CI Pipeline from GitHub that we have setup for this purpose.

yarda, 2025-01-07:
I used `pytest -s -v tests` to run the tests. Bad thing is that it requires X server, but I think I could workaround it with the null X server on the build machine. Now, I run it manually. The numpy-1 was run on Fedora 41, the numpy-2 was run on Fedora 42 (rawhide), but the python versions should be the same.

numpy-1:
342 tests passed

numpy-2:
333 tests passed, 3 failed, 6 skipped due to coredump

So it isn't compatible, but it isn't so bad.

yarda, 2025-01-07:
numpy-1 warnings:
```
====================================================================================== warnings summary ======================================================================================
tests/auto_interpretation/test_auto_interpretation_integration.py: 1 warning
tests/auto_interpretation/test_center_detection.py: 1 warning
tests/auto_interpretation/test_message_segmentation.py: 1 warning
tests/auto_interpretation/test_noise_detection.py: 1 warning
tests/test_filter.py: 6 warnings
  /usr/lib64/python3.13/site-packages/urh/signalprocessing/Signal.py:210: DeprecationWarning: Python 3.14 will, by default, filter extracted tar archives and reject files or modify their metadata. Use the filter argument to control this behavior.
    obj.extract(members[0], QDir.tempPath())

tests/awre/test_address_engine.py: 2 warnings
tests/awre/test_awre_real_protocols.py: 2 warnings
tests/awre/test_generated_protocols.py: 9 warnings
tests/test_fuzzing_profile.py: 1 warning
tests/test_maincontroller_gui.py: 1 warning
tests/test_project_manager.py: 5 warnings
tests/test_protocol_label_dialog.py: 3 warnings
tests/test_signal_tab_GUI.py: 1 warning
tests/test_simulator_tab_gui.py: 1 warning
  /usr/lib64/python3.13/site-packages/urh/signalprocessing/ProtocolAnalyzer.py:733: DeprecationWarning: Testing an element's truth value will always return True in future versions.  Use specific 'len(elem)' or 'elem is not None' test instead.
    if not root:

tests/awre/test_address_engine.py: 5 warnings
tests/awre/test_awre_real_protocols.py: 8 warnings
tests/awre/test_generated_protocols.py: 15 warnings
tests/test_fuzzing_profile.py: 2 warnings
tests/test_maincontroller_gui.py: 2 warnings
tests/test_project_manager.py: 9 warnings
tests/test_protocol_label_dialog.py: 3 warnings
tests/test_signal_tab_GUI.py: 1 warning
tests/test_simulator.py: 6 warnings
tests/test_simulator_tab_gui.py: 8 warnings
  /usr/lib64/python3.13/site-packages/urh/signalprocessing/Ruleset.py:143: DeprecationWarning: Testing an element's truth value will always return True in future versions.  Use specific 'len(elem)' or 'elem is not None' test instead.
    if tag:

tests/awre/test_address_engine.py: 1218 warnings
tests/awre/test_awre_preprocessing.py: 2057 warnings
tests/awre/test_awre_real_protocols.py: 340 warnings
tests/awre/test_checksum_engine.py: 40 warnings
tests/awre/test_generated_protocols.py: 1463 warnings
tests/awre/test_length_engine.py: 673 warnings
tests/awre/test_partially_labeled.py: 410 warnings
tests/awre/test_sequence_number_engine.py: 139 warnings
tests/test_analysis_tab_GUI.py: 712 warnings
  /usr/lib64/python3.13/site-packages/urh/awre/Histogram.py:102: DeprecationWarning: tostring() is deprecated. Use tobytes() instead.
    values[vector[start : start + length].tostring()].append(i)

tests/awre/test_address_engine.py: 46 warnings
tests/awre/test_awre_real_protocols.py: 22 warnings
tests/awre/test_generated_protocols.py: 51 warnings
  /usr/lib64/python3.13/site-packages/urh/awre/engines/AddressEngine.py:534: DeprecationWarning: tostring() is deprecated. Use tobytes() instead.
    result[p1].add(val.tostring())

tests/awre/test_address_engine.py: 46 warnings
tests/awre/test_awre_real_protocols.py: 22 warnings
tests/awre/test_generated_protocols.py: 51 warnings
  /usr/lib64/python3.13/site-packages/urh/awre/engines/AddressEngine.py:535: DeprecationWarning: tostring() is deprecated. Use tobytes() instead.
    result[p2].add(val.tostring())

tests/awre/test_address_engine.py: 530 warnings
tests/awre/test_awre_real_protocols.py: 55 warnings
tests/awre/test_generated_protocols.py: 217 warnings
  /usr/lib64/python3.13/site-packages/urh/awre/engines/AddressEngine.py:326: DeprecationWarning: tostring() is deprecated. Use tobytes() instead.
    if i in rng.message_indices and rng.value.tostring() in addresses

tests/awre/test_address_engine.py: 410 warnings
tests/awre/test_awre_real_protocols.py: 44 warnings
tests/awre/test_generated_protocols.py: 55 warnings
  /usr/lib64/python3.13/site-packages/urh/awre/engines/AddressEngine.py:357: DeprecationWarning: tostring() is deprecated. Use tobytes() instead.
[... 207 lines cut ...]
      raise RuntimeError("cannot join current thread")
  RuntimeError: cannot join current thread
  
    warnings.warn(pytest.PytestUnhandledThreadExceptionWarning(msg))

tests/test_simulator_tab_gui.py::TestSimulatorTabGUI::test_open_simulator_dialog_and_send_message
  /usr/lib/python3.13/site-packages/_pytest/threadexception.py:82: PytestUnhandledThreadExceptionWarning: Exception in thread Thread-35 (simulate)
  
  Traceback (most recent call last):
    File "/usr/lib64/python3.13/threading.py", line 1041, in _bootstrap_inner
      self.run()
      ~~~~~~~~^^
    File "/usr/lib64/python3.13/threading.py", line 992, in run
      self._target(*self._args, **self._kwargs)
      ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/lib64/python3.13/site-packages/urh/simulator/Simulator.py", line 248, in simulate
      self.process_message()
      ~~~~~~~~~~~~~~~~~~~~^^
    File "/usr/lib64/python3.13/site-packages/urh/simulator/Simulator.py", line 383, in process_message
      self.log_message("  Received {} data bits".format(len(received_msg)))
                                                        ~~~^^^^^^^^^^^^^^
  TypeError: object of type 'NoneType' has no len()
  
    warnings.warn(pytest.PytestUnhandledThreadExceptionWarning(msg))
```

yarda, 2025-01-07:
Except of the `AttributeError: 'NoneType' object has no attribute 'start'` which I currently have no idea where the problem is, the rest is probably caused by the `gauss_fir` in `src/urh/cythonext/signal_functions.pyx:241`. I guess it's related to the numpy-2 change in data type promotion, which is now according to the [NEP 50](https://numpy.org/neps/nep-0050-scalar-promotion.html#nep50), i.e. the `float_32` was promoted to `float_64` which maps to the C type `double`. Probably conversion function `.astype(np.float32)` on the right place should fix it?

yarda, 2025-01-07:
With the following patch:
```
--- a/src/urh/cythonext/signal_functions.pyx
+++ b/src/urh/cythonext/signal_functions.pyx
@@ -238,9 +238,10 @@ cdef np.ndarray[np.float32_t, ndim=1] gauss_fir(float sample_rate, uint32_t samp
                                                 int(filter_width * samples_per_symbol) + 1,
                                                 dtype=np.float32)
     cdef float ts = samples_per_symbol / sample_rate  # symbol time
-    cdef np.ndarray[np.float32_t] h = np.sqrt((2 * np.pi) / (np.log(2))) * bt / ts * np.exp(
+    cdef np.ndarray[np.double_t] h = np.sqrt((2 * np.pi) / (np.log(2))) * bt / ts * np.exp(
         -(((np.sqrt(2) * np.pi) / np.sqrt(np.log(2)) * bt * k / samples_per_symbol) ** 2))
-    return h / h.sum()
+    cdef np.ndarray[np.float32_t] hh = h.astype(np.float32)
+    return hh / hh.sum()
 
 cdef float clamp(float x) nogil:
     if x < -1.0:
```
tests results: 1 failed, 341 passed

The patch is far from being optimal any hints for optimization are welcome.

Regarding the remaining 1 failure, it's in:
```
self = <tests.awre.test_generated_protocols.TestGeneratedProtocols testMethod=test_without_preamble>

    def test_without_preamble(self):
        alice = Participant("Alice", address_hex="24")
        broadcast = Participant("Broadcast", address_hex="ff")
    
        mb = MessageTypeBuilder("data")
        mb.add_label(FieldType.Function.SYNC, 16)
        mb.add_label(FieldType.Function.LENGTH, 8)
        mb.add_label(FieldType.Function.SRC_ADDRESS, 8)
        mb.add_label(FieldType.Function.SEQUENCE_NUMBER, 8)
    
        pg = ProtocolGenerator(
            [mb.message_type],
            syncs_by_mt={mb.message_type: "0x8e88"},
            preambles_by_mt={mb.message_type: "10" * 8},
            participants=[alice, broadcast],
        )
    
        for i in range(20):
            data_bits = 16 if i % 2 == 0 else 32
            source = pg.participants[i % 2]
            destination = pg.participants[(i + 1) % 2]
            pg.generate_message(
                data="1010" * (data_bits // 4), source=source, destination=destination
            )
    
        # self.save_protocol("without_preamble", pg)
        self.clear_message_types(pg.messages)
        ff = FormatFinder(pg.messages)
        ff.known_participant_addresses.clear()
    
        ff.run()
        self.assertEqual(len(ff.message_types), 1)
    
        mt = ff.message_types[0]
        sync = mt.get_first_label_with_type(FieldType.Function.SYNC)
        self.assertEqual(sync.start, 0)
        self.assertEqual(sync.length, 16)
    
        length = mt.get_first_label_with_type(FieldType.Function.LENGTH)
>       self.assertEqual(length.start, 16)
E       AttributeError: 'NoneType' object has no attribute 'start'

tests/awre/test_generated_protocols.py:85: AttributeError
```
and here I have no idea what's wrong, any hints?

If we come up with complete solution, I am OK to create PR.

yarda, 2025-01-07:
Cleaner patch:
```
--- a/src/urh/cythonext/signal_functions.pyx
+++ b/src/urh/cythonext/signal_functions.pyx
@@ -238,9 +238,9 @@ cdef np.ndarray[np.float32_t, ndim=1] gauss_fir(float sample_rate, uint32_t samp
                                                 int(filter_width * samples_per_symbol) + 1,
                                                 dtype=np.float32)
     cdef float ts = samples_per_symbol / sample_rate  # symbol time
-    cdef np.ndarray[np.float32_t] h = np.sqrt((2 * np.pi) / (np.log(2))) * bt / ts * np.exp(
+    cdef np.ndarray[np.double_t] h = np.sqrt((2 * np.pi) / (np.log(2))) * bt / ts * np.exp(
         -(((np.sqrt(2) * np.pi) / np.sqrt(np.log(2)) * bt * k / samples_per_symbol) ** 2))
-    return h / h.sum()
+    return (h / h.sum()).astype(np.float32)
```

yarda, 2025-01-07:
The proposed patch requires check for numpy version, because it will fail with numpy-1, e.g. something like:
```
    if np.lib.NumpyVersion(np.__version__) >= '2.0.0b1':
        ...
    else:
        ...
```
But I am not sure how to write it in cython.

andynoack, 2025-01-08:
Thank you very much for you work so far! Do you like to open a pull request? I will add Johannes to this issue for reviewing.

yarda, 2025-01-08:
Yes, I will open PR when I found a solution. There are still two things I need to resolve:
1) add numpy version conditional into `signal_functions.pyx` for the patch to work with numpy-1 and 2. Unfortunately, I am not good at cython, so all my attempts how to conditionalize it (nicely) so far failed.
2) I have no idea what's wrong with the test_generated_protocols.py, the traceback is not very useful and I am now digging into the `FieldType.Function.LENGTH` code, but my progress is slow.

yarda, 2025-01-09:
I probably resolved the 1., PR will follow. Still fighting with the 2.

yarda, 2025-01-09:
The 2. seems to be some problem in the `FormatFinder.run()`.

yarda, 2025-01-09:
The problem seems to be in the SequenceNumberEngine.

andynoack, 2025-01-16:
Can we mark this issue as resolved?

yarda, 2025-01-16:
> Can we mark this issue as resolved?

Unfortunately, it isn't resolved yet. There is still numpy-2 related failure of the `test_generated_protocols.py`. I am still on it, but there is no traceback that could help and I am not familiar with the code and with the math used. But I am slowly progressing. There was also a bit delay, because I focused on the wrong thing - it uses sets and then iterates over them and it seems on Fedora 42 the sets are internally sorted differently than on Fedora 41 which gives different bit dumps. But it turned out this is not the core of the problem, the core problem is some change in numpy-2 which I hasn't been able to identify yet. I also had to work on different more urgent problem, but I think I will be able to resolve it.

PR #1176 (2025-06-19): Feat!: use numpy2
Fix type problem with numpy2.
Drop support for numpy1 to simplify testing.

## Reference

fixes #1149
closed 2025-06-19 by andynoack