jopohl/urh #1158: How to edit a recorded signal before replaying it?

gitdode, 2025-03-22:
I can record the signal of the transmitter and replay it to the receiver - which is just awesome. Also demodulating the signal, reading the payload etc. works great - fantastic!

Now I'd like to change just a few bits in the payload of the recorded signal, but I don't see how to do it.
I tried to generate a signal based on the recorded signal but did not succeed at all.

What am I missing? Is it possible at all what I am trying to do?

andynoack, 2025-03-23:
You can delete some passages from your raw signal, copy and paste stuff or even add a custom sine wave. But if you want to modify your signal based on bits/bytes, you need to use the generator. Here, you try to find a modulation that is the same or very similar to the original and then edit the bits in the signal you create from zero or a a drag and dropped signal.

gitdode, 2025-03-25:
Thank you for your swift reply!
It took me a while to figure it out, but it works - very nice!